OrcaSlicer/OrcaSlicer #6058: Raft not working with multiple objects

regularlabs, 2024-07-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

14.5 (23F79)

### Printer

Bambu Lab P1S

### How to reproduce

- Place 2 objects on the print plate
- Set raft layers to 2 or 3 or whatever
- Auto-arrange object, so that they have the minimum distance
- Slice plate 

### Actual results

- Serious warning about gcode conflicts
- Looking at the rafts top layer, it creates overlapping paths

### Expected results

The raft should be generated correctly

### Project file & Debug log uploads

[Raft conflict.zip](https://github.com/user-attachments/files/16123849/Raft.conflict.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

regularlabs, 2024-07-08:
This seems somewhat connected:
https://github.com/bambulab/BambuStudio/issues/2273

regularlabs, 2024-07-09:
Issue seems to be caused by some check whether the "Enable Support" option is on.
If it IS on, then the auto-arrange will give the objects enough space for the individual rafts.
If it is NOT on, the rafts will overlap.

But like also mentioned in the https://github.com/bambulab/BambuStudio/issues/2273, the rafts shouldn't be made object dependent. If you have 20 items close to each other, it should just create 1 single raft for all objects.

DeadMedicKampfer, 2024-09-25:
I'm not having this same problem, but I'm having trouble figuring out how to put down a raft for better support on large prints.

regularlabs, 2024-09-25:
@DeadMedicKampfer What is the use of referencing a completely different question in this issue?

kingnoob1377, 2024-10-01:
[image]
Picture of overlapping raft. This seems like a obvious feature missing rather than a bug.

regularlabs, 2024-10-01:
This seems like a bug, rather than a missing feature. 😜

regularlabs, 2024-10-07:
[image]
That nearly broke my printer, as OrcaSlicer was telling it to print in the same spot like 8 times because of overlapping rafts.
I had to first 'Assemble' all the small parts to work around this major issue.

Ergonomicmike, 2024-12-26:
Still an isssue in v2.3.0-dev, Dec 25 build.

[image]

inspectionsbybob, 2025-01-15:
I have the same issue, but with ANY object, so there is a conflict with something.  I tried it with just a cilindar primative, and got the error.
I am Win11, Orca Slicer 2.2.0
Enclosed are the Error and my 3MF 
[image]
[Cylinder with Raft.zip](https://github.com/user-attachments/files/18417405/Cylinder.with.Raft.zip)

Hope this helps

closed 2025-04-23 by github-actions[bot]