OrcaSlicer/OrcaSlicer #6059: Align and distribute objects

Alkerion, 2024-07-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It's a nightmare to align parts, especially with the awful and unusable coordinate system.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add buttons to align and distribute selected objects.

Left, right, top, bottom, middle, center, ... 
Distribute on X, on Y
Place on bed

kylek29, 2024-07-08:
>It's a nightmare to align parts, especially with the awful and unusable coordinate system.

Parts in general or do you mean things like modifiers? Because I absolutely agree on the modifiers (e.g. cutters, etc.) -- I don't think that feature received the love it needed to support a proper workflow. Just posting this here for discussion purposes ..

Possibly bringing over some QoL elements from CAD would help? Such as:

- Ability to change a model -or- sub model (model within assembly) origin point (right now, I think everything defaults to center bottom?).
- Ability to "snap to vertex" move an object. So you'd highlight. model and hold a modifier key (or use a UI toggle?) which shows the most prominent vertexes (e.g. the corners), you select one then you repeat by selecting another models vertex, it'd move the model.
- Snap to Edge (same as above)
- Snap to Face (same as above)
- Ability to "lock" to those ^ , so if the other part is manipulated, they stay attached to the selection parts.
- Introduce Assembly Coordinates, so when a model is apart of an assembly the assembly uses the world coordinates and each part is fixed to the internal coordinate system (so 0,0,0 of the assembly on the plate is not the same as 0,0,0 for the parts that make up that assembly). It may already do this? I haven't tested.

Alkerion, 2024-07-08:
Absolutely everything as the coordinate system is just unusable.

A slicer is not a CAD software, so we cannot really compare IMHO.

Total non sense to have the Z at the middle of the part for a slicer, the Z at the bottom is the only right choice.

Same non sense to have always the origin on the first plate on a multi plates project.

Amazing that these issues still not being fixed.

closed 2024-10-15 by github-actions[bot]

itsgudenuf, 2026-03-27:
Just curious if the Align/Distribute menu options that Bambu Studio 2.5 can be ported to Orca?