OrcaSlicer/OrcaSlicer #6064: Appimage Fails On Debian

dener3id, 2024-07-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Debian Sid XFCE Latest

### Additional system information

Ryzen 2700x 64 gb ram AMD RX 580

### Printer

N/A

### How to reproduce

Run appimage under Debian and not Ubuntu.

### Actual results

Doesn't open when launched it is marked as an executable other appimages run.

### Expected results

Runs splash screen then launches App

### Project file & Debug log uploads

N/A crashes before even creating the folder in .config

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dener3id, 2024-07-09:
I'll report back in the next two or three days on if it has the same issue on deb stable.

nils-a, 2024-07-09:
Probably related to #5635

dener3id, 2024-07-09:
> Probably related to #5635

Is there a way to test that? I'm happy to run that also having a similar issue with freecad if that helps (I can run other appimages example renderdoc). I didn't get an error output like that guy did unless I'm looking in the wrong place.

nils-a, 2024-07-09:
> Is there a way to test that?

I started the AppImage on the console, thus having some output on the console:

```sh
9:39:08 ❯ ./Applications/OrcaSlicer_Linux_V2.1.1.AppImage 
/tmp/.mount_OrcaSl9cFMpK/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

```

dener3id, 2024-07-10:
> @nils-a See #4616 for that specific issue TLDR: It's resolved in the latest nightly build

Where is the daily build I'm having trouble finding the page?

dener3id, 2024-07-11:
> > Where is the daily build I'm having trouble finding the page?
> 
> https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds Use the AppImage with `Ubuntu2404` in its name

Thanks seems to work so far. A .deb release would limit this kinda headache.

dener3id, 2024-07-11:
If someone wants help getting auto build to work with .deb,aur etc I'd be willing to help.

serhii-nakon, 2024-07-14:
@dener3id Try `sudo apt install libwebkit2gtk-4.0-37`

serhii-nakon, 2024-07-14:
@dener3id Hello I use Debian 12 and all works fine, please start application from console to see error
@nils-a Hello, maybe you does not have `libwebkit2gtk-4.0-37` try `sudo apt install libwebkit2gtk-4.0-37`

nils-a, 2024-07-15:
@serhii-nakon I, personally, am not running debian - I simply came here looking if someone had the same problem as I had.
Yes, installing the correct dependencies should fix the problem. However, the main point of providing an AppImage is that nothing should need to be installed.

But like @ElectricalBoy said: The newest nightly fixed the problem. No need to install anything else.

kwagnitz, 2024-07-28:
OrcaSlicer_Linux_V2.1.1.AppImage runs OK for me on Ubuntu 24.04.
But the icon that lands on the dock is the gear icon.

From https://github.com/TheAssassin/AppImageLauncher/issues/653 :
the OrcaSlicer desktop file is incorrect. They need to specify their StartupWMClass so that the window manager can associate the running program's windows with the pinned entry.
~> xprop WM_CLASS
WM_CLASS(STRING) = "orca-slicer", "OrcaSlicer"
Thus, the desktop file needs to be amended with either
StartupWMClass=orca-slicer
# alternatively
StartupWMClass=OrcaSlicer

References:
    https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html
    https://specifications.freedesktop.org/startup-notification-spec/startup-notification-0.1.txt

kwagnitz, 2024-07-28:
Hopefully the first line was of some use.  Does run on a Debian derivative.

#6276

closed 2024-11-03 by github-actions[bot]