OrcaSlicer/OrcaSlicer #6065: Support for displaying time and filament usage on printer screen

hamzademirr, 2024-07-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

### Description
Currently, Orca Slicer adds time and filament usage information in a format that some 3D printers cannot interpret correctly. For example, it uses:
; filament used [mm] = 1182.01
; filament used [cm3] = 2.84
; filament used [g] = 3.53
; filament cost = 14.10; total filament used [g] = 3.53
; total filament cost = 14.10
; total layers count = 137
; estimated printing time (normal mode) = 23m 46s

However, many printers expect this information in the following format:
;TIME:209.287
;Filament used:0.334524m
;Layer height:0.2

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

### Suggested Solution
Please provide an option in the slicer settings to choose the format of the time and filament usage information. This would help users whose printers require different formats.

Noisyfox, 2024-07-09:
Which printer uses this format more speifically?

queeup, 2024-09-19:
For example Creality 3 V3 SE.

antonini, 2024-11-23:
Same issue over here. With Elegoo Neptune 4 Pro

queeup, 2025-01-21:
Maybe we can add `Placeholders` for `{estimated_printing_time}` & `{filament_used_mm}` so we can put this placeholders to our  gcode.

```
;TIME: {estimated_printing_time}
;Filament used: {filament_used_mm / 1000}
;Layer height: {layer_height}
```

hamzademirr, 2025-01-21:
I tried to solve it on gcode for a long time but I did not get any results. I am a v3 se user. When I had a problem with marlin, I turned to the klipper installation and indirectly solved this problem.

rgembalik, 2025-01-23:
I'd love to see at least placeholders for this. I use several estimation-related plugins in OctoPrint, but octo & cost estimation (whichever does the calculation) plugin do not understand the comments produced from orca.

eduard-sukharev, 2025-03-01:
I am contributor to Ender 3 V3 SE fork and would love to see it implemented. Currently Marlin parses first N rows of gcode to get meta info and scanning through multi-megabytes file to get that metadata would add a huge delay upon opening the file. Also, the printing time is already split into units, but Cura/Prusa/Creality slicers print them as seconds - it would be nice to have it in seconds too.

I'm really not sure how (and if) other printers handle this gracefully, but having this basic metadata at the top should probably do good for them too.

protoz, 2025-03-17:
After switching to Orca, the Spoolman plugin for Octoprint no longer works as Octoprint cannot pull the filament information from the file. Having that variable would likely fix my issue and be the last thing I needed from Cura.

davydmaker, 2026-02-05:
Until Creality updates the firmware, you can try installing a custom one like this: https://github.com/navaismo/Ender-3V3-SE
It has this fix as well as several other interesting ones.

But in addition to that, I've released an enhancement to OrcaSlicer that allows adding an initial header to the G-code, allowing you to include:
```
;TIME:{print_time_sec}
;Filament used:{filament_length_m}m
;Layer height:{layer_height}
```

Additionally, I created these placeholders/variables that provide access to the data in the format expected by Marlin-based firmware.

I'm asking for your support here: https://github.com/OrcaSlicer/OrcaSlicer/pull/12186