OrcaSlicer/OrcaSlicer #6069: [OrcaSlicer 2.1.1] Segfault when going to machine's g-code twice

brunetton, 2024-07-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Ubuntu 23.10

### Additional system information

Intel Core i5-8250U 64 / RAM 16 GiB / Intel UHD Graphics 620

### Printer

Personalized version of Ender 3

### How to reproduce

1. open OrcaSlicer
2. open printer configuration
3. goto g-code
4. close printer configuration
5. open printer configuration again
6. goto g-code

### Actual results

the app crashes, segmentation fault in logs

### Expected results

I should be able to visualize and edit printer's G-code

### Project file & Debug log uploads

[Untitled.zip](https://github.com/user-attachments/files/16142658/Untitled.zip)

```
[error]	2024-07-09 13:23:15.695937[Thread 0x0000712fe05db380]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-07-09 13:23:15.712983[Thread 0x0000712fdd1246c0]:Unable to open the file/tmp/.mount_OrcaSl7Eg1XY/resources/calib/tolerance_test/OrcaToleranceTest.stl
[warning]	2024-07-09 13:23:16.427503[Thread 0x0000712fe05db380]:01:23:16 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-09 13:23:16.497883[Thread 0x0000712fe05db380]:01:23:16 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-09 13:23:16.502055[Thread 0x0000712fe05db380]:01:23:16 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-09 13:23:16.611137[Thread 0x0000712fe05db380]:01:23:16 PM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-07-09 13:23:18.743566[Thread 0x0000712fe05db380]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-07-09 13:23:18.815036[Thread 0x0000712fe05db380]:get_version, get_version not supported,return 00.00.00.00!
[error]	2024-07-09 13:23:20.262547[Thread 0x0000712fe05db380]:calc_exclude_triangles:Unable to create exclude triangles
```

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

- installed via app image (last release:  [OrcaSlicer_Linux_V2.1.1.AppImage](https://github.com/SoftFever/OrcaSlicer/releases/download/v2.1.1/OrcaSlicer_Linux_V2.1.1.AppImage) https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.1.1)
- maybe related to https://github.com/SoftFever/OrcaSlicer/issues/5030#issuecomment-2092386058 ?

timnolte, 2024-07-11:
I believe this is a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/6020

henrivdr, 2024-07-12:
This could be a duplicate, but my reproduction steps are closer to this one.

serhii-nakon, 2024-07-16:
I have the same issue

serhii-nakon, 2024-07-16:
For me last working stable version is 2.0.0, and not working 2.1.0 and newer

serhii-nakon, 2024-07-17:
Looks like it related with new Mesa due my friend use 2.1.1 with Debian 12 and main differences are - mesa, kernel, desktop environment

kernel: my - 6.9.7 (git), friends - default Debian's 6.1.99
mesa: my - 24.1.2 (git), friends - default Debian's 22.3.6
DE: my - Gnome, friends - KDE (both default Debian's)

And that are all differences between our systems.

JooJooBee666, 2024-07-21:
Same here.  2.1.1 just seg faults when loading any model.
KDE 6.1, X11, Latest Mesa
AMD Radeon RX 6700 XT (radeonsi, navi22, LLVM 18.1.6, DRM 3.57, 6.9.9-200.fc40.x86_64) (0x73df)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.3.0-devel

Tried older versions, but the result is the same, segfault right after opening.  Version 2.0.0, 2.1.0 and 2.1.1 app images.  Very sad today. :(

serhii-nakon, 2024-07-21:
> Looks like it related with new Mesa due my friend use 2.1.1 with Debian 12 and main differences are - mesa, kernel, desktop environment
>
> kernel: my - 6.9.7 (git), friends - default Debian's 6.1.99 [...]

I just checked with older Mesa and issue still exist, also I tried with Xorg and Wayland - still exist, and looks like one main difference between my system and my friends is DE or list of libraries.

@SoftFever Where I can find debug symbols for latest builds? To get more information why it segfault in my case. I found that latest build with debug symbols was 2.0.0 but for me sqgfault starts from 2.1.0

JooJooBee666, 2024-07-23:
I's working again for me after updating to Plasma 6.1.3 and Mesa.

AMD Radeon RX 6700 XT (radeonsi, navi22, LLVM 18.1.6, DRM 3.57, 6.9.9-200.fc40.x86_64)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.3.0-devel
Release      : 0.3.20240723.10.8f9a157.fc40

closed 2024-10-29 by github-actions[bot]