OrcaSlicer/OrcaSlicer #6072: Prepare Display Blacked/non-functional

bennejam000, 2024-07-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win 11 23h2

### Additional system information

CPU: AMD Ryzen 7 5800X 8-Core Processor                4.20 GHz
RAM: 32.0 GB Corsair 3200MHz
MOBO: Asus TUF x570 +wifi
GPU: AMD 6700XT  - Driver: 24.6.1

### Printer

heavily modified ender3 on klipper

### How to reproduce

1. Open Orca Slicer 2.1.1 or 2.1.0
2. start new project with empty plate
2b. open previous project (doesn't matter which one)

### Actual results

Prepare tab plate display is black

### Expected results

Should display empty plate and when adding new stl or when opening previous project, should display plate with previous project

### Project file & Debug log uploads

[debug_Tue_Jul_09_08_19_33_43592.log.zip](https://github.com/user-attachments/files/16143929/debug_Tue_Jul_09_08_19_33_43592.log.zip)
[debug_Tue_Jul_09_08_17_12_53460.log.zip](https://github.com/user-attachments/files/16143931/debug_Tue_Jul_09_08_17_12_53460.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

found this when researching the issue. https://www.reddit.com/r/BambuLab/comments/1dtgvve/black_screen_in_prepare_tab_of_bambu_slicer/

it's Bambu's slicer, but hopefully some helpful info.

bennejam000, 2024-07-11:
Update: another user posted on that r/BambuLab thread. They stated turning off a bunch of features specifically for their program "fixed" the issue. I took the time to go setting by setting and narrowed it down. The Radeon Image Sharpening enabled causes the black screen. Disabling just for Orca Slicer and no more black screen. It's a solid work around/fix. If this doesn't help with codebase, I hope it at least helps others as they search for a quick fix.

bepstein111, 2024-07-11:
I'm on linux and am having a similar issue. Unsure if related. Same CPU but NVIDIA 3070Ti for GPU, debian linux for OS. One day I opened up OrcaSlicer and the prepare tab is just gray, I can't see anything in the 3D view, but I can see all the settings on the left. Once I switch to the Preview tab, the GCode preview loads fine. I can open a new issue if you'd prefer.

amcdannell2002, 2024-08-02:
> Update: another user posted on that r/BambuLab thread. They stated turning off a bunch of features specifically for their program "fixed" the issue. I took the time to go setting by setting and [...]

Can confirm that this worked!!!

bepstein111, 2024-08-02:
> Can confirm that this worked!!!

Unfortunately, I don't think this helps those of us with NVIDIA GFX. I'll play around with the nvidia control panel and see if similar settings helps me out.

noone606, 2024-09-01:
Having the same problem after a windows update this morning 
Lenovo slim 5 pro AMD Ryzen 7 7840HS with Radeon 780M Graphics &  RTX 4060
eventually worked out if iI add ORCA to desktop apps in
 windows >display>graphics
and force to use the nVidia card problem went away
seems this overrides the settings in nVidia control panel now

eZtaR1, 2024-10-01:
Ryzen 7 5800X
6700 XT

Did this in AMD adrenalin and then it worked after restarting orcaslicer.
[image]

DIJIANYUE, 2024-12-24:
关闭HDR即可解决！
if you close HDR？

closed 2025-04-01 by github-actions[bot]