OrcaSlicer/OrcaSlicer #6073: Feature request: Merge in "No perimeters on bridge areas" from SuperSlicer

tastyratz, 2024-07-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Bridging issues with Arachne are prevalent where it constantly tries to draw a bridge over free air.  While this setting won't SOLVE many of those edge cases it WILL allow bridges to properly anchor in many other scenarios and drastically improve bridge performance.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think Orcaslicer could adopt this already available feature from SS for big gains in bridging performance.

### Additional context

A similar request was made upstream at Prusaslicer
https://github.com/prusa3d/PrusaSlicer/issues/12721
However, there are no signs yet that they are considering any kind of pull back.

lhndo, 2024-08-24:
Related: https://github.com/SoftFever/OrcaSlicer/issues/5376

tastyratz, 2024-11-23:
I really hate to see this one go stale because it's such a big issue with an existing workaround we can port from superslicer. Fixing Arachne upstream is a bigger conversation but maybe this is a quick fix that helps for now?

ZigZagT, 2024-12-07:
Hit this issue as well. This is a major problem for functional parts, where thick walls are often required for strength.

In my case, My print is effectively capped at 3 wall loopss (.82mm line width w/ .8mm nozzle). Orca slicer starts to generate floating walls and bridges with 4 & above wall loops.

*3 wall loops*
[image]

*4 wall loops*
[image]

henrivdr, 2024-12-07:
I absolutely upvote this one as well. It also causes weird behavior where the ceiling of the benchy cabin is printed completely in mid air, instead of using a bridge.

evilC, 2024-12-14:
I would also like to see this, but for a different reason
I am currently trying to print in CarbonX ezPC+CF, and this stuff is a right pain with bridging
For normal bridging, I can slow right down and it will be OK, but not with the perimiters
Here is a test part:
[image]
Here is how it comes out:
[image]
As you can see, the normal bridge is OK, but the perimiters simply do not hold up. The bottom right is the START of the bridge perimeter.
What I think is happening is that for the perimeter, the pressure has built up in the nozzle, and once the floor beneath the nozzle disappears, the filament is free to just squirt out.
When it does the normal bridge, it will stop extruding, move to the start of the bridge, and then start extruding, and I think it must not have that same pressure built up.
I have tried using "Extrusion Rate Smoothing", even setting it to a ludicrously low value of `2`, to get it to ramp down the pressure really slowly, and start slowing down earlier:
[image]
But this does not solve the problem.
My hunch is that if I could disable bridge perimeters, then all would be OK.