OrcaSlicer/OrcaSlicer #6074: Force cooling for overhangs and bridges doesn't work

vadixidav, 2024-07-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 7950X

### Printer

Bambu Lab A1

### How to reproduce

1. Print the attached file.

### Actual results

Regardless of what the fan speed is set to, the fans will not speed up during the overhang bridge or the bridging section. Check the first bridging section, for instance. You can try changing the settings however you like, I have tried no walls, all walls, etc but I cannot get the fan to speed up during bridging consistently. It will occasionally raise the fan speed, but it is rare, and it seems that the command to raise the fan speed is often omitted.

### Expected results

The fan should go to 80% when an overhang wall or bridge is encountered.

### Project file & Debug log uploads

[Bridging_Test_50 10% fan -> 80% fan keep on](https://github.com/user-attachments/files/16145504/Bridging_Test_50.zip)
[Bridging_Test_50 0% fan -> 80% fan keep off](https://github.com/user-attachments/files/16145524/Bridging_Test_0_80.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I am trying to print totally transparent prints (this is a cube, so very thick):
[image]

To do this, I can't have the fan on, but I have to have the fan on to bridge. Unfortunately, I have been unable to get the fan to speed up during bridging consistently. Often times, it will start up then slow back down to the normal speed after it has been bridging for about a second or two. The bridges are happening quite slowly at 1mm/s. The bridges come out absolutely perfect when the fan is on, but when it is not on they have various issues. Some settings do cause the G-code to be emitted to turn the fan up correctly, but again, it doesn't remain on very long. Usually it never even turns up.

vadixidav, 2024-10-08:
Still not stale. I have observed this countless times at this point in many other contexts. I now own an A1 Mini and a P1S in addition to the A1. The current situation is that the fan slows itself down to the normal speed when it is sped up for pretty much any reason, be it overhangs or bridges, but it only does so on the A1 and A1 Mini. The P1S does not have this issue, and the fan actually stays on during the entire bridge. This leads to significantly degraded bridging and overhang performance on the A1 and A1 Mini, as the fan can only speed up for very short sections. This happens regardless of whether the fan is in "keep on" mode or if it turns on from being off. It only happens on the A1 and A1 Mini.

vadixidav, 2024-12-08:
This is still an issue. The P1S always speeds up the fan and turns the fan off when it is supposed to. For some reason the A1 and A1 Mini G-code generators aren't properly operating the printer's fan controls.

closed 2025-03-16 by github-actions[bot]

makeitrio, 2025-05-08:
still problem. Brige fan doesnt works!

jackcummings, 2025-07-01:
This indeed is still a problem.  A1 Mini will consistently shut down the cooling at 100% for bridges and overhangs, always before the actual overhang or bridge is finished.  I am constantly fighting curling corners because of this