OrcaSlicer/OrcaSlicer #6078: Wiki enhancement - Wiki Pull Request

brunetton, 2024-07-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I printed pressure advance calibration lines but didn't noticed any difference between 0 and 1. So I jumped onto gcode and noticed that the code is making use of `M900` command, that is not compiled in my firmare.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I took time to clone the repo and its wiki and I added a mention to check that M900 is supported by printer's firmware: you can see it [in my fork's wiki](https://github.com/brunetton/OrcaSlicer/wiki/Calibration#pressure-advance)

You can cherry-pick the last wiki commit in my fork if you find this mention usefull

neo-jayfeather, 2024-07-14:
This was added [yesterday](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration/_compare/2ba1de3c6255213308a533c25185af7b3f714ce0), and should be resolved.

brunetton, 2024-07-16:
> This was added [yesterday](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration/_compare/2ba1de3c6255213308a533c25185af7b3f714ce0), and should be resolved.

Great :)

Closing