OrcaSlicer/OrcaSlicer #6079: WiFi connection to Qidi X-Plus printer cannot be established.

Uberschmutzer, 2024-07-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Qidi X-Plus

### How to reproduce

Prepare/Printer
Select Qidi X-Plus Printer, click on wifi.
Octo/Klipper, Put in printer IP, Click on Test-

### Actual results

 Get the following:
  
[image]

### Expected results

Should pass test and connect to printer.

### Project file & Debug log uploads

Never got here

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Lumesyn, 2024-07-15:
Is this the original Qidi X-Plus? Most Qidi printers don't have Klipper and a web-interface pre-installed, the WiFi function won't work without it. 

If you go to your printers IP address in your browser do you get a web interface? Some Qidi's are weird and utilize port 10088, try that as well.

Uberschmutzer, 2024-07-15:
Thanks for taking a look at this.  It is the original Qidi X-Plus.  With the Qidi Print and Control Panel applications I can connect with Wifi although it's always been problematic.  Ethernet has been more reliable. No luck with web interface or using port 10088.  I can always use the Control Panel app with OrcaSlicer to use this older printer.

closed 2024-07-15 by Uberschmutzer