OrcaSlicer/OrcaSlicer #6080: Compare Filament presets

Ergonomicmike, 2024-07-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

My current workflow: When I start with a new type of filament (PLA, PETG, etc.) I make a "Master" Filament preset for that type. This "Master" is necessarily based on the first finish (i.e., matte, shiny, translucent) and color filament roll I try. (E.g., Translucent Red PETG.) 

And then after my first attempt at a "Master" preset, I'll tweak it, creating newer versions, as I tune up the preset via temp and retraction tests.

So then I eventually have "Master PETG 070924."

I thought that I could use the same Filament preset for different colors/finishes across the same type and brand of filament. (E.g., Eryone Translucent Red PETG and Eryone Red PETG.) But I can't. If nothing else, I've found that I sometimes have to adjust temperatures for different finishes of filament. 

The problem is that when I learn something new and modify my current "Master" Filament profile, I can't remember what I changed when I open an older Filament preset for a different finish.

For example, I just found - after my Extruder chewed a gap into my filament - that my Master PETG profile has been wrong from the beginning. In the beginning I had increased Retraction Distance for PETG (from my base Retract Distance set in my Printer preset) in the Filament Settings Override. But I didn't know to increase the Travel Distance Threshold too. 

Okay, that problem is fixed. In my Master preset. But tonight I opened an old PETG preset and almost forgot to change the Travel Distance Threshold in that. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If Orca had a way to compare Filament presets, as is already does for Process presets, then I could compare my current Master to whatever previous custom Filament preset I had made.

I'm hoping that the existing "compare" code can easily be repurposed to do the same for Filament presets.

(And I would offer to make a PR if I knew how to code and how to work the github.)

timnolte, 2024-07-11:
There already exists the ability to compare Filament presets. If you use the Compare Presets tool you can select different filament presets and then even transfer some settings between them over to another preset and then save it.

Ergonomicmike, 2024-07-11:
Tnx for commenting.

Terminology is a bit confusing because under "Filament" in Orca are two settings. One is presumably "Filament," and the other is titled Process. I  can compare Process. I can't compare Filament, per the screen shot.
[image]

timnolte, 2024-07-11:
So actually this setup is actually a little deceiving. Even though the "Compare Presets" tool is on the "Process" navigator, the tool does comparison of more than just "Process". Once you launch the tool you can choose 2 different Filament presets, either User or System, and compare them and transfer individual configuration items.

[image]

Ergonomicmike, 2024-07-11:
Ahh... okay. Exactly what I was asking for. Thanks.

So, then, I wonder if there is a place to move the Compare presets icon to that would make it clear that it can compare both Process and Filament? Or duplicate it, and have it default to "Filaments" proper when clicked a duplicate icon next to Filaments?

I'll close this issue in a week or so.

closed 2024-07-16 by Ergonomicmike