OrcaSlicer/OrcaSlicer #6082: Calibration - Pressure Advance

RTGRDH, 2024-07-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

macOS Sonoma Version 13.6.7

### Printer

Two Trees Sapphire Pro (SP-3) Modified

### How to reproduce

Calibration -> Pressure Advance -> any Extruder Type -> PA Tower or PA Pattern

### Actual results

[image]
[image]

### Expected results

Selecting PA Tower, a tower is expected, although Orca Slicer crashes

Selecting PA Pattern, a pattern is expected after slicing. Only a square is shown. Sent to printer and only a square was printed, no pattern at all.
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16157480/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

RTGRDH, 2024-07-10:
It is the preview attached to the ticket.
I tried printing the sliced object, only a tiny square were produced.

Noxandrys, 2024-09-30:
Ran into the same issue. Changing the 'Print Sequence' setting to 'By layer' fixed it for me. Looks like 'By object' causes some issues.

RTGRDH, 2024-10-01:
> Ran into the same issue. Changing the 'Print Sequence' setting to 'By layer' fixed it for me. Looks like 'By object' causes some issues.

Tried it out and it worked! Thanks a lot! I'll close this ticket.

closed 2024-10-01 by RTGRDH