OrcaSlicer/OrcaSlicer #6083: Release-packages without version in filename

rootiest, 2024-07-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'd like to be able to link directly to the latest release, but the filenames have the version number so that is not possible without knowing the version number in advance of constructing the link.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great if the release packages, (particularly the AppImage) didn't include the version number, or there were duplicate packages included without the version number.

My reasoning for this request is that it is not currently possible to link to those packages without knowing the version number in advance.

I would like my dotfiles to be able to pull the AppImage directly as part of my initialization, but in this case I am not able as I cannot construct a hyperlink like:

https://github.com/SoftFever/OrcaSlicer/releases/latest/download/OrcaSlicer_Linux.AppImage

This fails because while `latest` does work, the filename must match, and the filenames used here include the version number.

Thank you for considering this request!

-Rootiest

Samueru-sama, 2024-07-15:
>I would like my dotfiles to be able to pull the AppImage directly as part of my initialization

Doesn't this work? 

```
version=$(curl -Ls https://api.github.com/repos/SoftFever/OrcaSlicer/releases/latest | sed 's/[()",{} ]/\n/g' | grep -oi "https.*linux.*mage$" | head -1)
wget $version
```

You can even pipe the `$version` variable to a file and compare it and that way you are not always pulling it.

rootiest, 2024-07-15:
> Doesn't this work?

Yeah that seems to work great, now I feel silly for thinking it couldn't be done so concisely.

Thanks @Samueru-sama!

closed 2024-07-15 by rootiest

Samueru-sama, 2024-07-15:
> > Doesn't this work?
> 
> Yeah that seems to work great, now I feel silly for thinking it couldn't be done so concisely.
> 
> Thanks @Samueru-sama!

Btw I took that line from [AM](https://github.com/ivan-hc/AM) which I think does everything that you are trying to do in your dotfiles. It automatically makes a updater script for orcaslicer as well that does the version comparison and all of that.

rootiest, 2024-07-15:
> Btw I took that line from [AM](https://github.com/ivan-hc/AM) which I think does everything that you are trying to do in your dotfiles. It automatically makes a updater script for orcaslicer as well [...]

Thanks, I will check that out!

I am using [chezmoi](https://www.chezmoi.io/) to manage the dotfiles so I plan to use that line to get the url and then apply it to my chezmoiexternals template which will handle caching for me so it's not re-downloading it every time.