OrcaSlicer/OrcaSlicer #6089: Change Filament (usually M600) adds an instance at the beginning of the print when it doesn't nee to.

GregGFlynn, 2024-07-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3

### How to reproduce

Any time you add a Change Filament command when you right click on the layer you want to add it to, it also adds the filament change gcode (usually M600) to the beginning of the print where it is not needed.  It does put it in the correct place, but also adds one to the beginning that is not needed:

G1 Y200 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0
G90
G21
M83 ; use relative distances for extrusion
M600     **<--- this M600 is not needed at the beginning of the print.**
;_FORCE_RESUME_FAN_SPEED
T0
M104 S215 ; set nozzle temperature
; filament start gcode

I remove it with a text editor before I print, but it doesn't need to be added in the first place.

### Actual results

You have to go through the filament change steps at the beginning of your print when it is not needed.

### Expected results

Filament changes should only happen at the layers selected (no need for one at the beginning of the print).
[Megalodon_Tooth_Display_4DOM.zip](https://github.com/user-attachments/files/16164909/Megalodon_Tooth_Display_4DOM.zip)

### Project file & Debug log uploads

[Megalodon_Tooth_Display_4DOM.zip](https://github.com/user-attachments/files/16164913/Megalodon_Tooth_Display_4DOM.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

RemyLebeau, 2024-07-14:
I was just coming to post this bug!  I also have this issue.

RemyLebeau, 2024-07-14:
I found a solution in another thread about this issue:

https://github.com/SoftFever/OrcaSlicer/issues/2893#issuecomment-1868263435

closed 2024-10-20 by github-actions[bot]

larroy, 2025-01-27:
Please don't close this, the bug is still happening in 2.2.0 with multiple extruders, if you print in the second extruder an spureus M600 is inserted

harrislapiroff, 2026-01-04:
I also still have this issue!