OrcaSlicer/OrcaSlicer #6097: Incompatibility with older davinci printers

Bexin3, 2024-07-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hello, I am currently using a davinci 1.1 plus, and I as able to generate code and get it working on it from slic3r using a simple script that removes empty lines and adds some info - https://github.com/chrislondon/Convert_GCode_To_3W script but standalone, however for an some reason this doesn’t work with Orca slicer. The files as such are not accepted but after removing some of the images and such I was able to put it on the printer. However it wasn't entered and didnt seem to be printing.

Too left edge by deleted

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Would be great to see this fixed or an outright possibility to generate .3w code

Bexin3, 2024-07-11:
Specifically here you can see how the file looks when imported int XYZware from orca compared to slic3r
[image]
[image]

Bexin3, 2024-07-11:
Weirdly enough no, all behave the same

discip, 2024-07-12:
@Bexin3
Have you tried disabling **`Arc fitting`**?

[image]

Bexin3, 2024-07-21:
@discip 
Hello, sadly no luck
I had it disabled already, so tired enabling it, and while no luck, it seems to make it worse;
[image]

closed 2024-10-28 by github-actions[bot]