OrcaSlicer/OrcaSlicer #6098: Large item, large bed, Orca stalls when slicing at 60% but no indication of problem or crash.

tlhintoq, 2024-07-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win10x64

### Additional system information

[image]

### Printer

Creality CR10s-5

### How to reproduce

Large model with quite a lot of overhang.
Slice it with tree supports (any style and combination)
[image]

### Actual results

At 60% `Slicing plate `: Support: propagate branches at layer` the job seems to just stop slicing.  
There's no crash, the program still responds etc. but there's no indication that anything is actually happening.
Task Manager show no load, no demand beyond the same levels seen at idle time.
Let the job sit and continue to process just in case... 2 hours later there's no sign of any change.
[image]

### Expected results

1 - The progress dialog should be wider to tell the user *which* layer is being processed, instead of cutting off after the world `layer`.  At least then even if it were slow it would still be seen at layer 1500... 1501... 1502... and the user would know its still working
2 - If a subsystem doing the work has crashed there should be some notice of that... some watchdog process to know that no progress is taking place.
3 - It would be nice if it actually finished the job.  

It worries me that if a single large part on a 500mm³ machine is causing this kind of stall then having multiple parts or larger parts on a larger machine such as an Elegoo Giga (800x800x1000) are going to be impossible to slice.

### Project file & Debug log uploads

[Back.zip](https://github.com/user-attachments/files/16171366/Back.zip)
[debug_Thu_Jul_11_10_01_17_4524.log.zip](https://github.com/user-attachments/files/16171384/debug_Thu_Jul_11_10_01_17_4524.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-07-11:
Its worth noting this same part, same orientation and layerheight... same everything... slices fine if you use `normal` support style.  But at a difference of 21 days versus about 12 for tree support, and a 5x the cost due to amount of material, its not really an option I want to take.
[image]

Dirkkro, 2024-07-13:
Have same probllem, 
Version 2.1.1 
Windows 10, 
but old I5, 6500T 
6GB Ram

tlhintoq, 2024-07-13:
@Dirkkro 
FYI:  
PC reboot - Did nothing, launched nothing else, went straight to Orca.
Changed to Tree: Strong
And it sliced.

I feel regardless of the cause of the fail, magic combination of settings, conflict with another running program, variable array overrun if too many tree branches... whatever the actual cause... there needs to be at least 2 things:

1. A watchdog running to determine when the slicing process just evaporates and nothing is being done
2. The status dialog needs to be wider so user can see the full message including layer number so a human can also see when the process has just stopped.

tlhintoq, 2024-07-20:
@SoftFever 
Just updating... I can get a model to slice with `Tree(auto)|Strong` maybe 15% of the time.  I can't find a pattern, rhyme or reason to it other than it always drops off in this `60% propage branches at layer` point with no idea what layer its working on or if its working at all.

This is a real shame as the strong supports are wonderful, robust and often use less material and print faster.  I'd love to be using these full time.

I have some serious reservations about this though. If Orca can't handle larger items on a 500mm machine then I have very little hope for using it with my Elegoo OrangeStormGiga when it arrives.  Please please please can someone look in to this and not force me back to Cura!
If I had to **guess** - If its at branch propagation and mostly on larger items... then maybe its over running a `collection` of branches with more elements than it can handle.  ???
[image]

tlhintoq, 2024-07-20:
If it helps investigating - the ONLY change made between the above fail and this success was changing the style from `Strong` to `Default`.  Then it slices fine.
[image]
[chest105.orca.zip](https://github.com/user-attachments/files/16318390/chest105.orca.zip)
[log.zip](https://github.com/user-attachments/files/16318394/log.zip)

tlhintoq, 2024-07-20:
@Noisyfox 
So some news on this: I sent it slicing and just walked away for the day while taking care of other things.  I figured _"What the heck, costs me nothing to let it spin for a day."_
The file shown in the original/first post - the giant back armor
4 **minutes** - Sliced at `Tree(Auto)|Default` - Print time 10d 14hours
5 **hours** - Sliced at `Tree(Auto)|Strong` - Print time 7d 21hour

So this may not be that it fails - just that for some reason `strong` takes 200x more time to process - and the status box is so narrow you can't see if its working and making process because you can't see a report of the layer its working on.

3 days faster print time and more robust supports are definately an amazing goal, but it cost more than half a shift for the computer to slice that, compared to under 5 minutes for the same job on a different style of tree.  
### **That can't be right, can it?**

[image]

tlhintoq, 2024-08-04:
cc: @Noisyfox 

Still a problem with 2.1.1 nightly.
24 hours to slice something when using `Tree: Strong` yet no more than 10 minutes when using any other combination.

Its really a killer when trying to do larger items.  24 hours to slice makes it very prohibitive to try new options.  If you notice something not being supported for example and you have to go paint on a little more support: Bam that's an entire 24 hours again.

If there is anything I can do... anything I can provide... name it... to help facilitate this ticket just let me know.

tlhintoq, 2024-08-27:
No change with 2.2.0-dev nightly of 27aug2024

Noisyfox, 2024-08-27:
The strong tree algorithm is somewhat computational & memory extensive while unable to be parallelized, and I haven't find a way to improve it yet.

tlhintoq, 2024-08-27:
> unable to be parallelized

Wow.  Ok.  That explains it. Thanks for that.
Is there any chance of widening that status box (lower right corner) so that more information is visible and updating?  That would at least let users know the program isn't frozen/hung.  Right now it cuts off at `blahblah at layer:` then you can't see the layer number.  A little bit wider and we'd at least see the layer number incrementing.

Noisyfox, 2024-08-28:
> Right now it cuts off at `blahblah at layer:` then you can't see the layer number. A little bit wider and we'd at least see the layer number incrementing.

Good point, I will try doing that.

closed 2024-12-03 by github-actions[bot]