OrcaSlicer/OrcaSlicer #6101: Changing layer filament sequence causes slicing error

ycarriger, 2024-07-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab X1 Series

### How to reproduce

1. Start a new project.
2. Ensure there are at multiple filaments available.
3. On a blank plate, right-click and select Add Primitive, then select Cube.
4. In the left-hand Process section, select the Objects tab.
5. Click on the plate name, for the "Other layers filament sequence" option, change it from Auto to Customize (no need to actually change the default order).
6. Click the Slice Plate button.
7. An Orca Slicer error pops up, "Serializing NaN".

### Actual results

An Orca Slicer error pops up, "Serializing NaN"
[image]

### Expected results

Slicing completes successfully!

### Project file & Debug log uploads

[Orca Bug 1.zip](https://github.com/user-attachments/files/16173065/Orca.Bug.1.zip)
[debug_Thu_Jul_11_01_56_18_33788.log.zip](https://github.com/user-attachments/files/16173066/debug_Thu_Jul_11_01_56_18_33788.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

CallmeBigfat, 2024-07-27:
+1,  shame this issue happens.

ycarriger, 2024-08-18:
Changing the number for the layer range end layer or changing it to End will work around the issue, per workaround posted in Bambu Studio repo [here](https://github.com/bambulab/BambuStudio/issues/4511#issuecomment-2270322781)

CallmeBigfat, 2024-08-21:
> Changing the number for the layer range end layer or changing it to End will work around the issue, per workaround posted in Bambu Studio repo [here](https://github.com/bambulab/BambuStudio/issues/4511#issuecomment-2270322781)

thank you for this heads up, it did fix the issue for me :)

closed 2024-11-27 by github-actions[bot]