OrcaSlicer/OrcaSlicer #6105: configure step for 'dep_Boost' fails

superfloh247, 2024-07-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

latest master 45224ee

### Operating System (OS)

macOS

### OS Version

14.5

### Additional system information

build_release_macos.sh -t 14

### Printer

Bambu P1P

### How to reproduce

build_release_macos.sh -t 14

### Actual results

Performing configure step for 'dep_Boost'
-- Boost: using system layout: include, bin, lib, lib/cmake
-- Boost: Release build, static libraries, MPI OFF, Python OFF, testing OFF
-- Boost: libraries excluded: contract;fiber;numpy;stacktrace;wave;test
-- Boost.Context: architecture arm64, binary format mach-o, ABI aapcs, assembler gas, suffix .S, implementation fcontext
-- Boost.Iostreams: ZLIB ON, BZip2 ON, LZMA ON (multithreaded), Zstd ON
-- Boost.Locale: iconv ON, ICU OFF, POSIX ON, std ON, winapi OFF
-- Boost.Math: standalone mode OFF
-- Boost.Multiprecision: standalone mode OFF
-- Boost.Thread: threading API is pthread
-- Configuring done (1.0s)
CMake Error at libs/mysql/CMakeLists.txt:59 (target_link_libraries):
  The link interface of target "boost_mysql" contains:

    OpenSSL::Crypto

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

CMake Error at libs/numeric/odeint/CMakeLists.txt:15 (target_link_libraries):
  The link interface of target "boost_numeric_odeint" contains:

    Boost::mpi

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

CMake Error at libs/redis/CMakeLists.txt:49 (target_link_libraries):
  The link interface of target "boost_redis" contains:

    OpenSSL::Crypto

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

-- Generating done (0.5s)
CMake Generate step failed.  Build files cannot be regenerated correctly.

### Expected results

build succeeds

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-07-13:
Duplicated with [#5916](https://github.com/SoftFever/OrcaSlicer/issues/5916)

closed 2024-07-13 by SoftFever

superfloh247, 2024-07-16:
howtio downgrade cmake to 3.29.6:

brew unlink cmake

wget https://raw.githubusercontent.com/Homebrew/homebrew-core/1976f46fc84ea7716722a067c0dcffb072a38388/Formula/c/cmake.rb

brew install -s ./cmake.rb