OrcaSlicer/OrcaSlicer #6106: Printer view for FlashForge printers

JMcrafter26, 2024-07-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

A printer view page that shows information about the printer (FlashForge Printers).
Currently, the page shows a “page access denied” error, and that is not optimal.
I found an API for FlashForge printers and tried it and it works.

Here is the API I am talking about: https://github.com/01F0/flashforge-finder-api

I would use the API (or the code logic) to get the printer details, such as print status. 
To get the camera (e.g., on the AD 5M Pro) I would use the printer this URL scheme: http://PRINTER_IP:8080/?action=stream

The good thing about the API is, that it works with many FlashForge printers such as Adventurer 5M, Adventurer 4, Finder and maybe more.

### Additional context

Example API response:

```JSON
{
  "Firmware": "v2.7.2",
  "Name": "Adventurer 5M Pro",
  "SN": "XXXXXXX",
  "Tool Count": "1",
  "Type": "Flashforge Adventurer 5M Pro",
  "X": "220 Y: 220 Z: 220"
}
```

```JSON
{
  "BytesPrinted": 11,
  "BytesTotal": 100,
  "PercentageCompleted": 11
}
```

?action=stream page:
[image]

timnolte, 2024-07-11:
I wonder if the Printer View in Orca actually has it's own interface setup? From what I've seen it seems like it's just a feed from the web interface of a printer. Perhaps it is using an API for other printers. Ideally, Orca-Forge would just contribute back to OrcaSlicer their work to integrate the printer interface view.

JMcrafter26, 2024-07-11:
> Ideally, Orca-Forge would just contribute back to OrcaSlicer their work to integrate the printer interface view.

This would be the best solution, but I don’t think they will do it since they added a login in order to use the device view and they also completely rebranded the slicer

JMcrafter26, 2024-07-12:
I just made a simple demo on how you could use the api with html and python. Here is the repo I made:
https://github.com/JMcrafter26/flashforge-printer-preview

Please note that this is just a demo and not nearly done

snintendog, 2024-08-01:
Would love to Access the Onboard models as well like FlashPrint has as well.

timnolte, 2024-10-31:
It would be nice to see someone be able to build a screen with some sort of API connectivity to the printers. Perhaps that is just beyond what is possible.

Rebel98luder, 2025-01-12:
Would love this fixed.  My biggest complaint while using ORCA with my Flashforge adventure 5m(non pro)

Goto the "device" tab, And this is all i ever can get. 

[image]

bstott, 2025-04-21:
I guess not enough of us complain or simply don't care. I would like to have this feature working on my printer.

timnolte, 2025-04-21:
Is this really an OrcaSlicer issue? Seems this is a FlashForge issue because they don't provide a proper network web interface for the AD5M machines. I don't believe OrcaSlicer ever provides some custom coded device control they just present the printer network web interface within the Device tab. If you put the Klipper mod on the AD5M, which gives you real Klipper instead of FlashForge's limited version, you will get full control in OrcaSlicer.

snintendog, 2025-04-21:
Given someone already hacked together an External solution to act as a bridge for Orca and FF.. Hell FF made their own FORK at this point. I doubt its Flashforge's fault and more the Orca devs not willing to do anything with Flashforge. Why should i CFW my machine to let Orcadevs pat themselves on the back for a good day of no work. After they Tossed away Support in general to just give it all to Bambuu's Proprietary software I am losing Faith in the Devs here. Almost nothing there has made it here. updates are nill to null. Barely any fixes.

timnolte, 2025-04-21:
I'm sorry but FlashForge, or any other manufacturer, doing their own proprietary garbage doesn't make it the OrcaSlicer Devs fault. Put the blame on the right place. FlashForge didn't need to fork OrcaSlicer but they did instead of just supporting OrcaSlicer. You can get what you want with this feature request by just using the FlashForge fork.

snintendog, 2025-04-21:
So In otherwords Litterally a guy not part of Orca figured it out using python... yet orca can't? I am sorry should i Giva A crap when the FLIPPING Api is OPEN and Orca ignores it for a YEAR now? And i am supposed to be angry that flashforge made a fork after Orca sat on their butt? This is supposed to be a opensource program focused on as many printer working at ONCE in the program... well I do feel like its Orcas fault for doing Jack all despite EVERY THING BEING HANDED TO THEM ON A SILVER PLATTER.

snintendog, 2025-04-21:
> Entitled much? How much are you paying the OrcaSlicer developers for your request. Oh right nothing. I also don't see a pull request with the completed code if it's so easy to just make it happen. Oh yeah, and how many other printers does OrcaSlicer already support and how many features do they already have to support. Oh, and how much is everyone else paying the Devs for fixing anything else or adding new features. Again nothing.

So its my job to Program this.... on a Issue board. On a project I DON'T MANAGE.Get blocked.

bstott, 2025-04-21:
snintendog - who are you ranting at?   I see you have done nothing on github.   Hope you weren't bashing me.  I haven't even gotten used to Orca to work well with it. I simply noted a feature seems a big deal - But - doesn't work.

bstott, 2025-04-21:
Also - timnolte - Too bad you feel so bad.  I would put in - IF you work Orca and IF I cared.  However, ORCA has this large Tab/Page for connecting to printers - doesn't work with mine.  I don't see any docs related to config files I have access to for getting my FlashForge Ad 5M to connect.  Like a Klipper or other FW related setup.   I simply chimed in it would be a nice feature - IF it was working - and other programmers haven't seemed inclined to upgrade.  To you:  Pound Sand. I wouldn't want your help - even though you are quite active.  Now that you've shown as a petulant child.   Your pissed attitude stinks as a person.

bstott, 2025-04-21:
snintendog - Needs to go piss in someone else's pool.    He's gone mental.

timnolte, 2025-04-22:
> So its my job to Program this.... on a Issue board. On a project I DON'T MANAGE.Get blocked.

@snintendog where is your pull request with the required code changes to implement this. I don't understand what your "job to Program this" is even talking about. Seems like you are the one pissing in the Orca pond and have no clue how open source software works. You get something for free with no investment other than making demands and calling out people that give of their own personal time when companies like FlashForge take advantage of the contributions of the community for their own financial gains. If you need this feature and it's your job then go to FlashForge and make your demands. Why don't you just you Orca-Flashforge which has what you want instead of being entitled.

@bstott

snintendog, 2025-04-22:
> snintendog - who are you ranting at? I see you have done nothing on github. Hope you weren't bashing me. I haven't even gotten used to Orca to work well with it. I simply noted a feature seems a big deal - But - doesn't work.

Nope not you @bstott . it was aimed at Timnolte. its been since 2023 since Flashforge Opened up the API to the AD5/Pro and since 2024 when they made Flash Orca their Answer to orca Ignoring Flash Forge Users since the beginning.. ANd i am talking about OP having already basically makeing a Local server to relay. I only have stated before that its been Forever since Orca abandoned FF completely there is a FeckTon of issues in 2.20 and 2.30 with FF printers ignored.

Also @timnolte I ONLY POINTED OUT WHAT IS IN THIS THREAD ALREADY! You hate the fact i AM correct.

timnolte, 2025-04-22:
@snintendog you don't get it. The Orca Devs aren't the only ones supporting every printer under the son. It's a community and manufacturer effort. Even Elegoo themselves did the work that is required to support the new Centauri Carbon printer by opening up a Pull Request with the required code changes to support their printers. In what fantasy world do you live in to somehow expect other people to support hardware and printers they don't even own. As I said you don't get how open source software works and are out of touch with your requests.

snintendog, 2025-04-22:
> [@snintendog](https://github.com/snintendog) you don't get it. The Orca Devs aren't the only ones supporting every printer under the son. It's a community and manufacturer effort. Even Elegoo [...]

Already Exists Twice over. Orca has always had a chip on its should over Flashforge and obviously you do too. Flash Orca was an Old Push that Orca kept ignoring and only forced FF to do SOMETHING on their own. and OP made a MOD to Orca that works and even THEN Orca refuses to implement ANYTHING Flashforge Related. I DONT OWN ORCA Why are you Telling me to Repeat History at this point.

timnolte, 2025-04-22:
@snintendog also I have been using OrcaSlicer with my FlashForge AD5M since it was supported in the 2.0 development release and have had no problems. I don't know what things you feel are majorly broken. I have over 3200 hours in the last 11 months using it with OrcaSlicer.

DreamingLamb, 2025-04-29:
> Would love this fixed. My biggest complaint while using ORCA with my Flashforge adventure 5m(non pro)
> 
> Goto the "device" tab, And this is all i ever can get.
> 
> [image]

I'm not home to find the setting, but there's a spot in OrcaSlicer to add the device page, it needs to be the http://PRINRET_IP:8080/?action=stream instead of just the IP.  

All I've managed to get from it is yhe Camera feed though.  I assume there's another page I cannot find that Orca-flashforge uses.

zweidrache, 2025-10-08:
Flashforge printers do appear to have a Web UI, as I discovered not through their fork of Orca, but rather specifically their dedicated monitoring software [https://github.com/Parallel-7/FlashForgeUI-Electron](https://github.com/Parallel-7/FlashForgeUI-Electron). when opening the software, you can go into Settings to view the port and set a password to access it, and then load it in a browser via the printer's IP address and that port (i.e.: http://PRINTER_IP:3000/).

[image]

[image]

However, when I tested this myself, despite it saying "Connected" at the top, trying to utilize any controls does not work -- but it's also broken in Flashforge UI proper, so I'm thinking I have to mess around with my local network settings to allow the printer to connect the way this intends to see the printer, as I have already set up via the FF login and have been using the printer fine without FFUI for the last month. I also can't figure out how to make that URL usable inside OrcaSlicer.

But I want to move to OrcaSlicer eventually because of features that haven't been implemented into Orca-FF yet and potentially install custom firmware later on, so being able to see the Web UI and send files directly to the printer instead of saving them to send through another app or transfer via USB would be ideal. I hope this information is helpful.

timnolte, 2025-10-08:
@zweidrache are you sure this actually turns on a WebUI on the printer. If it did you would be able to access it from a standard web browser. If you have a camera attached then you can use that port to view the camera only. I really don't think TF presents any WebUI on the Adventurer 5M printers beyond the camera.