OrcaSlicer/OrcaSlicer #6113: Wifi connection drops big files then go to print

chilito, 2024-07-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

win 10

### Printer

Flashforge M5

### How to reproduce

1. Send big file (more 50 megs) to print
2. Looking to UPLOAD messege 
3. Got ERROR

### Actual results

Got error and abort print

### Expected results

You need to increase the time to send large files or with a slow connection, at least up to 1 minute, but it’s better to call and choose how long to wait if the Wi-Fi is weak or got large file 

### Project file & Debug log uploads

any big file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dan4ik123654, 2024-07-12:
I also faced this problem

psihop, 2024-07-12:
Yes. I confirm. The problem exists

RusLevel, 2024-07-12:
I confirm. There is such a problem.

firiat345, 2024-07-12:
I also have the same problem.

mortonlame, 2024-07-12:
Yes. I confirm. The problem exists

Bato360, 2024-07-13:
Yes, this problem has existed since version 1.9, so it is not possible to use this slicer with the AD5M printer. And the modified weight from FF is generally based on the old version of the slicer 1.8

AceAsket, 2024-07-17:
Same problem

Volkov-Stanislav, 2024-07-17:
same problem too

hellysu, 2024-09-07:
Same problem

WebBasedUser, 2024-09-24:
Issue still exist in 2.2.0. Any solutions?

closed 2025-01-01 by github-actions[bot]