OrcaSlicer/OrcaSlicer #6114: Color paint by line range (Horizontal) leaving 2 colores on same layer

ericgs, 2024-07-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

Artillery Hornet whit Klipper and mods. using m600 as color change command

### How to reproduce

Import stl model whit angled surfaces, like the https://www.printables.com/model/178035-cute-mini-octopus/files, in exampled i have used the straight v6 suport model,

add color change by layer height range horizontal, set to 5mm for example, click the base of the model to paint from bottom.

[image]

slice model using 0.2mm layer

### Actual results

multiple color change commands inserted at layer.

[image]

### Expected results

single color at given layer 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

ericgs, 2024-07-12:
[Octopus_sup_v6.zip](https://github.com/user-attachments/files/16197367/Octopus_sup_v6.zip)
proyect file, sorry

ericgs, 2024-07-12:
may be related to [https://github.com/SoftFever/OrcaSlicer/issues/6089](url)

JasonHorkles, 2024-10-11:
Rude

SrGatoo, 2024-10-22:
Can confirm it's happening. Same procedure, same results with another file.

closed 2025-01-28 by github-actions[bot]

kt0mmi, 2025-02-09:
Having the same issue.