OrcaSlicer/OrcaSlicer #6123: Strange artefacts when slicing model

Surion79, 2024-07-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win 11

### Additional system information

AMD CPU, AMD GPU 64 GB RAM

### Printer

Voron 2.4R2

### How to reproduce

get file from Galileo 2 repo "[a] xl_stepper_plate_right x2.stl" or respective left. and lisce it

### Actual results

[image]

### Expected results

[image]

This (expected result) has been sliced with superslicer 2.5.59.12

### Project file & Debug log uploads

[debug_Sat_Jul_13_10_26_52_7844.log.zip](https://github.com/user-attachments/files/16201442/debug_Sat_Jul_13_10_26_52_7844.log.zip)

[Issue_Slicing.zip](https://github.com/user-attachments/files/16201445/Issue_Slicing.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Ergonomicmike, 2024-07-13:
related? https://github.com/SoftFever/OrcaSlicer/issues/5288

Surion79, 2024-07-14:
nope, it is not a single line, the additional layer/heigt is ~50% and not single lines.
also it is not a pure rendering issue. it really prints that way

YozenPL, 2024-08-20:
I also have some strange behavior with top visible layer.
BambuStudio:
[image]

OrcaSlicer:
[image]

This is exact the same 3mf file (I removed some letters to not make any promotion)
File: [logo_part.zip](https://github.com/user-attachments/files/16674012/logo_part.zip)

closed 2024-11-26 by github-actions[bot]