OrcaSlicer/OrcaSlicer #6124: Flow Rate Pass 1 & 2 should set "Top shell thickness" and "Bottom shell thickness" to 0 to avoid a solid print

awabom, 2024-07-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Sovol SV08 (Based off Voron 350 preset)

### How to reproduce

1. Set top shell thickness to 1 mm, set bottom shell thickness to 1 mm (Pretend we have a preset using these values)
2. Press 'Calibration' - Flow Rate - Pass 1
3. Slice plate

### Actual results

Inspect layers - notice all layers are 100% solid instead of the usual result where layer 2 and 3 have sparse infill.

### Expected results

If top/bottom shell thickness is 0, I get 2nd and 3rd layer using sparse infill to avoid having the first layer squish affect the top layer results. This should happen automatically when using the Flow Rate calculation function.

### Project file & Debug log uploads

[BugReport - Flowrate Test - Pass1.zip](https://github.com/user-attachments/files/16202031/BugReport.-.Flowrate.Test.-.Pass1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

wbagnall, 2024-07-14:
I second this. Flow rate calibration is inaccurate with solid objects, as the squish from the first layer builds up over subsequent layers, causing a false "over extrusion" on the top layer. 

[Ellis' Print Tuning Guide](https://ellis3dp.com/Print-Tuning-Guide/articles/extrusion_multiplier.html) goes into more detail about this. 

>  This small bit of infill helps to decouple the first layer squish. Otherwise, an over-squished first layer can propagate all the way to the top - making it appear overextruded.

jeremytodd1, 2024-07-14:
Whenever I use Orca's flow rate calibration tests I scale up the z just a tad so there is at least a couple layers of infill in it. This way, any first layer squish will cancel out throughout the infill layers.

awabom, 2024-07-15:
> Whenever I use Orca's flow rate calibration tests I scale up the z just a tad so there is at least a couple layers of infill in it. This way, any first layer squish will cancel out throughout the infill layers.

The user should not need to modify anything to make the test work (within reason). That's the reason I created this Issue.

jeremytodd1, 2024-07-15:
> The user should not need to modify anything to make the test work (within reason). That's the reason I created this Issue.

Oh, I fully agree that the user shouldn't need to do anything. I was just saying that there is a second solution to this issue that the developers could implement as well, if they have a preference.

closed 2024-10-22 by github-actions[bot]