OrcaSlicer/OrcaSlicer #6135: Precise wall creates inconsistent width on small cylindrical features

tomoinn, 2024-07-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

6.5.0-41-generic #41~22.04.2-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun  3 11:32:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### Printer

RatRig vcore 3.1 300, 0.6mm nozzle

### How to reproduce

Slice a vertical cylindrical tube with OD 10mm ID 8mm, enabling precise wall. 0.5mm line width. The attached project file has this as a protrusion above a wider part (the wider part does not exhibit any issues)

### Actual results

An inconsistent diameter cylinder is sliced, with some regions that have two walls and some with one. When printed this results in an uneven vertical cylinder with some areas having a larger diameter than others

### Expected results

The cylinder should have the same wall count at all layers and be uniform in diameter through its entire length. Disabling precise wall produces this result with two walls on every layer

### Project file & Debug log uploads

[smoker_lid_project.zip](https://github.com/user-attachments/files/16228335/smoker_lid_project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Ataraxia-Mechanica, 2024-07-15:
Precise wall made your wall a bit thinner to prevent overlapping, and this triggered arachne to change it to a single wall. Make the minimum width lower changes it to double walls:

[image]

tomoinn, 2024-07-17:
Thanks - that helps in this case, but it seems odd the behaviour would change over Z for an otherwise constant diameter part? Is this some kind of rounding error kicking in, or cumulative error tipping over a point where the wall algorithm determines it needs to go to a single thickness?

closed 2024-10-24 by github-actions[bot]