OrcaSlicer/OrcaSlicer #6136: Issue with Z-hop Types During Retraction to the Purge Tower in 3D Printing

amzaldua, 2024-07-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Printer

Bambu Lab A1 series

### How to reproduce

1.	Load the 3mf program file attached.
2.	Set the Z-hop type to ‘normal’ in printer profile/extruder settings.
3.	Start the print and observe the behavior when the printer moves to the purge tower after filament extrusion.
4.	Repeat the steps with Z-hop types set to ‘slope’, ‘spiral’, and ‘automatic’.
5.	Observe and compare the behaviors in each mode.

### Actual results

In normal mode, after finishing filament extrusion, the printer lifts the Z-axis before moving to the purge tower. In slope, spiral, and automatic modes, no Z-axis retraction occurs; the printer moves directly to the purge tower, grazing the print, and performs the retraction movement upon reaching the purge tower.

### Expected results

In all Z-hop modes, the printer should lift the Z-axis before moving to the purge tower to avoid grazing the print and ensure consistent print quality.

### Project file & Debug log uploads

[Honda keychain test.3mf.zip](https://github.com/user-attachments/files/16228477/Honda.keychain.test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

This issue is not observable during the preview of the print paths; it is only noticeable during the actual 3D printing process
The videos are too large. How can I provide them to you?

te7suo, 2024-07-15:
I have the same problem

DearTanker, 2024-09-15:
I have a similar problem, Orca doesn't seem to use retraction before wiping, because Orca thinks it doesn't need to retraction after wiping. Bambu Studio will retraction, I will post an issue later to describe it in detail.

closed 2024-12-23 by github-actions[bot]