OrcaSlicer/OrcaSlicer #6137: "Access violation" when slicing a single model with print sequence set to By object

ReimuTheMiko, 2024-07-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11 23H2

### Additional system information

CPU: Intel Core i9-11900k 8 core
Memory: 4 x 16 RAM sticks
GPU: MSI RTX 3060 Ventus 3x 12G 

### Printer

Anycubic Kobra (using a modified profile for the Kobra 2)

### How to reproduce

1. Open a single model (in my case, Warhammer 40k ethereal complete model) in an otherwise empty object
2. Verify the print sequence is set to "By object"  (Others > Special mode > Print sequence)
3. Enable supports
4. Click Slice plate

### Actual results

[image]

### Expected results

The model should slice with generated supports (like this screenshot but set to By object)

[image]

### Project file & Debug log uploads

[debug_Sun_Jul_14_11_57_25_13004.log.zip](https://github.com/user-attachments/files/16228512/debug_Sun_Jul_14_11_57_25_13004.log.zip)
[issue with printer.zip](https://github.com/user-attachments/files/16228513/issue.with.printer.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

buzzhuzz, 2024-07-22:
Somehow your 3fm file has "extruder=1" set for the model. Not sure why it happens because there are no this setting enabled with in the GUI and extruder be defaut to "0".
[image]

You could try to export model as STL, create the new project and import that STL back.

There are a number of assertions along the source code which could catch this problem, but they are disabled for release (non-debug) buids.

aaronlee2701, 2024-09-08:
I'm facing the same issue too. Any fix yet?

closed 2024-12-15 by github-actions[bot]