OrcaSlicer/OrcaSlicer #6145: Changing default bridging direction

DusekM98, 2024-07-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I feel like it would make more sense to have the default direction for bridging so they are made in the shorter way.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

At least check some orientations for possible best one before generation.

### Additional context

[image]

vovodroid, 2024-07-17:
Isn't this option you are looking for?

[image]

DusekM98, 2024-07-17:
I know about this option. I meant it as an improvement to the generative algorithm. I feel like checking at least a few angles (eg. increments of 15) and choosing the best of them shouldn't take that much more time during slicing.

DusekM98, 2024-08-20:
I had it on zero and it gave me the result posted above, so maybe some improvements could be made.

vovodroid, 2024-08-20:
And 180?

DusekM98, 2024-08-20:
180 overrides the angle as intended by this setting. But then it does not use the automatic direction optimization I was talking about.

closed 2024-11-26 by github-actions[bot]