OrcaSlicer/OrcaSlicer #6153: Disable retrarction for Hilbert pattern

vovodroid, 2024-07-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Recently I've got a heatbreak clog printing top surface Hilbert curve. With disabled retraction it worked fine. So because of prolonged low speed printing retraction could lead to clogs.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Disable retraction for Hillbert curve pattern.

discip, 2024-07-19:
@vovodroid
Since you are on the **`Hilbert Curve`**.

Could you also please implement:
1. an option to set the resolution for it (make the individual segments longer or shorter) and
2. make it so that it rotates every other layer by 90° when it is used in **`Internal solid infill pattern`**.

discip, 2024-07-19:
@vovodroid
Since you are on the **`Hilbert Curve`**.

Could you also please implement:
1. an option to set the resolution for it (make the individual segments longer or shorter) and
2. make it so that it rotates every other layer by 90° or even better by 45° when it is used in **`Internal solid infill pattern`**.

vovodroid, 2024-07-19:
> Since you are on the **`Hilbert Curve`**.

I'm not on it, it's feature request, so I hope somebody else do it))). Anyway:

> 1. an option to set the resolution for it (make the individual segments longer or shorter) and

I guess it depends on line with and can't be changed

>  when it is used in **`Internal solid infill pattern`**.

Any meaning to use Hilbert for solid?

discip, 2024-07-19:
> I'm not on it, it's feature request

Oh, sorry. Somehow I mistook it for a PR.
Could be due to the fact that you submitted quite a few recently.

> I guess it depends on line with and can't be changed

Most likely, but it would be great, if one could change that.

> Any meaning to use Hilbert for solid?

Yes, this could prevent warpage when using **`Extra perimeters on overhangs`**.
Mentioned by CNC Kitchen somewhere in [here](https://www.youtube.com/watch?v=B0yo-o47688&t=1s).

vovodroid, 2024-07-20:
> Most likely, but it would be great, if one could change that.

I don't think it could be changed, otherwise line will not fill all surface without gaps.

> this could prevent warpage when using **`Extra perimeters on overhangs`**.

This article about external post processing script generating infill on its own. Slicer has no link to this.

closed 2024-10-27 by github-actions[bot]