OrcaSlicer/OrcaSlicer #6154: Underextrude the first layer of the skirt so it can be easily removed

Ataraxia-Mechanica, 2024-07-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Skirt stuck to the bed, need scraping to remove

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a flow multiplier setting for the first layer of the skirt so it can be removed easier

vovodroid, 2024-07-16:
Or add skirt width, that essentially is the same. But if you has such strong adhesion you can reduce it in a few ways.

kisslorand, 2024-07-16:
...or use the existing "skirt height", make it more than 1 single layer (I use 3 layers for skirt height)

Ataraxia-Mechanica, 2024-07-16:
I have it on 2 loops and 5 layers and it's still too strong with PETG.

Luke56784, 2024-07-17:
If you underextrude the skirt you will probably have the problem that you cannot remove it in one piece because it will break.

If you are using a PEI sheet I would recommend switching to a fiberglass (G10 / FR4 / Garolite) plate for PETG.
Same for TPU.

This two filament types often sticking too well to the PEI surface.

vovodroid, 2024-07-17:
> I have it on 2 loops and 5 layers and it's still too strong with PETG.

And what about printed model? Try to reduce bed and first layer temperature,  clean the bed.

> If you are using a PEI sheet I would recommend switching to a fiberglass (G10 / FR4 / Garolite) plate for PETG.

No problems with textured PEI and PETG.

Ataraxia-Mechanica, 2024-07-17:
> If you underextrude the skirt you will probably have the problem that you cannot remove it in one piece because it will break.
Nah, just underextrude the first layer and keep other layers normal.

> And what about printed model? Try to reduce bed and first layer temperature, clean the bed.
Bed is cleaned and new, any lower and parts would warp.

Ataraxia-Mechanica, 2024-07-17:
I'm going to write a post-processing script just for myself if yall think this is unnecessary smh, just making a suggestion that might be useful

closed 2024-10-24 by github-actions[bot]