OrcaSlicer/OrcaSlicer #6155: Orca Slicer Profile for Flashforge Adventurer 4

FrankHartog, 2024-07-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have already tryed my self by adjusting an Adventurer 3 profile and copied some start en end code from a Fl;ashprint generated gx file. But am am stuck on flowrate and z-hop issues.
I have been able to print halfe a first layer so far.
My experience is limited on this kind of stuff.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

I woudl like a profile for teh Flashforge Adventurer 4

### Describe alternatives you've considered

1. Flashprint, but this programm doesn't have the great tunig and cilbration tools which really make a difference in print quality.

KArturITo, 2024-07-16:
I join the request, I also need such a profile.

LS240177, 2024-08-03:
Hallo, Ich benötige ebenfalls ein solches Profil

Sisane, 2024-08-14:
I join the request. It would be great to use Orca with Flashforge Adventure 4.

urself25, 2024-09-28:
I too join the request.

mjexcelle, 2024-10-26:
Hello, I would also like to see the Flashforge Adventure 4 added, especially since I see the 5M, 5M Pro, 3, Guider 3 and 2s have been already added.

lcoppenr, 2024-11-01:
Ditto.   Try on my own and it is beyond my ability

urself25, 2024-11-02:
[image]
It's coming. Got tired of waiting and learned how to do it.

lcoppenr, 2024-11-06:
Can't wait to see it.  If you need someone to test it, feel free to give me a shout.

urself25, 2024-11-06:
> Urself25. It's looking very promising. Is it available for download somewhere?

> Can't wait to see it. If you need someone to test it, feel free to give me a shout.

I will post it sometime this week or during the weekend. I'm having issues with the flow rate. I'm starting to wonder if it's not my printer the issue. My bowden tube has started to come off so i'm waiting for replacement to arrive to see if that will fix the issue.

urself25, 2024-11-07:
Ok, I think I have found the cause of the flow rate issue in Orcaslicer and it's with my printer. My feeding assembly is dirty and I need to clean it if not replace it. As such, I'll post the files sometime tonight or tomorrow for anyone to test and report issues.

I'll post the link here when ready.

urself25, 2024-11-08:
Ok. Feeding Assembly cleaned. Bowden tube collet changed and extruder/bed recalibrated.  Everything works very well. Made some flow rate (FR) tests using the included test in Orcaslicer with a FR at 0.94 based on what I could find in Flashprint. The FR was too high it seems. The FR that was the better looking was thee -20 and even that one had issues. All the other ones higher than that were under extruding probably due to the pressure build up.

[image]

I lowered the FR to 0.785 and I'm running the test again. Once the FR is dialed perfectly I'll finalize the presets and post the files and instructions soon after.

urself25, 2024-11-13:
Update: I was not able to tune de Flow Rate and had multiple print issues only to realize yesterday that my nozzle is the cause. There seems to be a crack between the assemble and the metal part of the nozzle and as such pressure can't build up leading to crappy extrusions. I changed the nozzle reran the test and was finally able to find the right FR (it's much higher than 0.785) and I also completed the Linear Advance (Pressure Advance) test. 

So, all that remains is finalizing the presets and I'll be able to post the files and instructions on how to install the profiles. My week is pretty full but I'll try to have it ready and posted by end of day Sunday. I'll post the link to the fork in this thread.

mjexcelle, 2024-11-13:
Just wanted to say your efforts and dedication to this issue is greatly appreciated! Thank you!

urself25, 2024-11-16:
Good morning!

Please find Version 0.1 of the Adventurer 4 profile for OrcaSlicer
https://github.com/urself25/OrcaSlicer_FF_AD4_Profile

It's a first release and my first release. Be kind. Report any issues and enjoy!

FrankHartog, 2024-11-18:
Dear Urself,

I have tried to install the AD4 in a clean Orca 2.2.0 install.
I am able to choos the AD4, but when I start a new project, the AD4 is not listed as default printer.
I did follow your instructions, but so far I am not succesfull yet.

Any advice?

Regards,

Frank

https://github.com/urself25/OrcaSlicer_FF_AD4_Profile

closed 2025-02-25 by github-actions[bot]