OrcaSlicer/OrcaSlicer #6158: OrcaSlicer randomly gets "Incorrect password" and disconnects printer + segfault

c---, 2024-07-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

Bambu P1S

### How to reproduce

This can not be reproduced reliably. It may have something to due with the printer losing network connectivity.

### Actual results

This only seems to have appeared very recently. I believe with the latest update of `libbambu_networking.so` and the Orca 2.1 release.

Occasionally the printer will become disconnected from OrcaSlicer. When attempting to reconnect it will sometimes work or sometimes say "Incorrect password" and require selecting the printer again then entering the security code (the code does not change, always the same).

Additionally sometimes during this sequence OrcaSlicer will freeze then segfault. Again, seems to have something to due with sporadic network connectivity as it seems to be crashing in `libbambu_networking.so`

### Expected results

The security code should continue to work until it's changed.

No crash.

### Project file & Debug log uploads

traps: orcaslicer_main[3335525] general protection fault ip:7224e6aa28d9 sp:72245b3f28d0 error:0 in libbambu_networking.so[7224e6901000+3a4000]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kwagnitz, 2024-07-17:
Me too.  P1S Ubuntu 23.04.

krmtech, 2024-07-20:
Me too on Mac OSX

dmayhem, 2024-07-28:
I see this very regularly. Orca Slicer 2.1.1, Fedora 40, Bambu A1

Vantage-DS, 2024-07-30:
Happening here too. MacOS, 2.1.1.

jxer, 2024-07-30:
glad its not just me win 2.1.1 both A series printers.

Spazdogs, 2024-08-03:
Just had this happen on OS X on an X1C, also LAN only mode.

jonwest, 2024-08-06:
Also happening for me, 2.1.1, OSX Intel.

c---, 2024-11-15:
After more testing this seems to be fixed as far as I can tell. I'm not sure exactly what fixed it, maybe a combination of things. It seems likely it was the firmware update.

Please reopen or comment if anyone continues to experience this issue.

jxer, 2024-11-15:
i still have the issue, please reopen

jimmiedave, 2025-01-07:
me too, but no apparent segfault.

Just occurred (twice this morning) after waking the Mac from sleep with Cura connected to printer locally via WiFi (not doing cloud printing at all - local network).

Cura was connected, watching a print, Mac went to sleep, some time went by, woke Mac and Cura now had "No Printer". Upon selecting the printer again, I got "Incorrect Password". Entered the unchanged password and Cura connected.

Orca 2.2.0, macOS Sonoma 14.7.1, M3 Max CPU, P1S firmware "01.06.01.02 (Latest version)", AMS firmware "00.00.06.44 (Latest version)"

jxer, 2025-01-07:
Do you mean Orca? Why are you saying cura?

jimmiedave, 2025-01-07:
> Do you mean Orca? Why are you saying cura?

Yes, Orca. Dumb mistake. Will edit.

jimmiedave, 2025-01-07:
I've pretty definitively nailed my experience down to **network connectivity loss**, not sleep.

1. turn off WiFi, connect Mac to LAN with Ethernet
2. Start a print, monitor it from Device tab.
3. Pull out Ethernet connection.
4. Get this message (which I haven't seen before, probably because the machine was asleep when it would've popped):
[image]
5. Video drops
[image]
6. Then "No printer"
[image]
7. Reconnect Ethernet.
8. Locate printer in the Printer section
9. Select for reconnect and (if I recall the fast sequence correctly): "Incorrect password" then app crashes and closes with no warning or dialog.
10. Upon re-opening Orca, visiting the "Device" tab, it finds the printer and connects without asking for the password.

yachal-upson, 2025-01-10:
Have been getting symptoms that seem very much like this (repeated loss of password, glitchy device status) for a few months. Feel like since Orca update. On V2.2.0. X1 firmware unchanged during period - that's sitting at the final version which supports X1Plus installation. 

Also experiencing issues with FTP timeout. Job loading is super hit and miss.

closed 2025-04-19 by github-actions[bot]