OrcaSlicer/OrcaSlicer #6160: A1 AMS Lite in OrcaSlicer says it Sup.PLA and not normal PLA.

williamrehndell, 2024-07-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

19045.4651

### Printer

Bambulab A1 + AMS Lite

### How to reproduce

I dont now if its possible.

### Actual results

PLA shows up as a support material when it’s regular PLA. It's possible to select like normal in Bambu Studio, but not in OrcaSlicer.
![orca-slicer_vu2f8thdzv]

### Expected results

Select normal material.

### Project file & Debug log uploads

[crash_Wed_Jul_10_23_43_06_0.log](https://github.com/user-attachments/files/16263328/crash_Wed_Jul_10_23_43_06_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

williamrehndell, 2024-08-18:
Tried loggin out and in, reseting the printers. Nothing solves the issue.

closed 2024-11-24 by github-actions[bot]

arushe0805, 2024-11-25:
I have the same issue.
Bambu Studio shows it as PLA but Orca as Support PLA:
[image]
[image]

Orca 2.2 and AMS Lite 00.00.07.94