OrcaSlicer/OrcaSlicer #6169: Orcaslicer forgets access code for X1C in LAN mode constantly

kk7ds, 2024-07-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5 (latest)

### Additional system information

Macbook Pro M1

### Printer

Bambu X1C (in LAN-only mode)

### How to reproduce

1. Open Orcaslicer
2. Go to device
3. Probably see that the printer in the unbound list
4. Click on it
5. Put in the passcode, see it move to the recent printers list
6. Close Orcaslicer, disconnect from the printer, or just let the laptop go to sleep so it disconnects
7. See that the printer as moved back to the unknown printers list, which will require a code again

### Actual results

Printer constantly goes to the unknown list requiring the access code when connecting.

### Expected results

Remember the access code. This worked fine before I updated to 2.1.1, continues to work fine in Bambu studio on the same machine with the same printer.

### Project file & Debug log uploads

No project file required. Can reproduce just while monitoring an existing print. Very frustrating during a multi-hour print that every time I wake the machine or go back into orca to check on the print, I have to dig out the access code again.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

markus-k, 2024-07-18:
Same behaviour on P1S

notreallylex, 2024-07-20:
Same behavior on A1 (01.03.00.00)

chilternflyer, 2024-07-21:
OMG I thought it was just me.. My P1S does the same thing.  
I've gone back to 1.9.1 and it all works fine.

SeriousSeri, 2024-07-23:
Same on my X1C with 2.1.1, Win11

dkoontz, 2024-07-24:
Same issue on my X1C also with v2.1.1, Win11

Zalatra, 2024-07-27:
I'm having the same issue, Orca Slicer 2.1.1, Windows 11. Uploads from slicer when pressing "Print Plate" have also slowed to a crawl - seemed to start at the same time.

peterloron, 2024-08-01:
Same here. Orca 2.1.1, MacOS 14.5 M1, X1C

Zalatra, 2024-08-03:
> I'm having the same issue, Orca Slicer 2.1.1, Windows 11. Uploads from slicer when pressing "Print Plate" have also slowed to a crawl - seemed to start at the same time. Sometimes the massive delay in uploading means the transfer fails.
> 
> When it does succeed, after taking 10-15 minutes to transfer a 3-5MB file for printing, the system consistently disconnects at the point where it switches to the "Device" tab. I have to reconnect (by reentering the access number) at that point, though the print starts successfully.

So... as far as I know I have changed absolutely nothing in my setup, but this has stopped happening. 🤷

kk7ds, 2024-08-03:
It occasionally doesn't happen for me. I'd say give it time :)

HodGreeley, 2024-08-05:
Same for me. A1 Mini, Orca 2.1.1, MacOS 13.4.1.

What does "give it time" mean?  It will randomly not require the code, but it's less than 1 in 5 times connecting, at best.  I've tried to figure out what makes the code stick, to no avail.  I can have my machine connect to watch a print, bring something else into the foreground for a while, and have to enter the code again.  Quite frustrating.

kk7ds, 2024-08-05:
@HodGreeley I said that because @Zalatra said it seemed to have "stopped happening". My point was "give it time, and I'm sure it will start happening again". Like you, it sometimes works as expected, but almost all of the time does not. I think @Zalatra was just experiencing one of those occasional works as expected spurts, so I was encouraging them not to think it had fixed itself.

NotLordCalvert, 2024-08-07:
Sam issue here. I used to experience this from time to time, but since updating to 2.1.1 it happens every single time my computer is brought out of sleep.

khlam, 2024-08-11:
On my Windows machine, setting "Network Profile Type" to "Private network" resolved my issue of the passcode being forgotten. I haven't found a solution for my Mac yet. YMMV

katherine-black, 2024-09-04:
The work around for this is to get all your bambu printers connected in Orca, close Orca. Open your `OrcaSlicer.conf` (`~/Library/Application Support/OrcaSlicer/OrcaSlicer.conf` on macOS), find the `"user_access_code"` section, and update it to be just `"access_code"`, save & close. Open Orca again, congratulations 🎉 , it won't forget your printers again.

Really this seems like a bug in the bambu network plugin more so than Orca, as the same failure is present in BS.

kk7ds, 2024-09-04:
It definitely is a bug in the plugin because it started happening with studio as well. It definitely seems like something Orca should be able to override, as even before this my printer would occasionally be forgotten.

Anyway, I don't have that section in my config at all:
```
$ grep access ~/Library/Application\ Support/OrcaSlicer/OrcaSlicer.conf
$
```
Can you show me what it should look like so I can add it?

HodGreeley, 2024-09-06:
@katherine-black fix works for me.  Conf file section looks like

` "access_code": {
        "serial number": "code"
    },`

Replace serial number with the hex serial number you can find under Device > Update.  Code is your 8 digit access code.

kk7ds, 2024-09-06:
@katherine-black @HodGreeley This doesn't seem to work for me. I had  no "access_code" section to copy. If I add a top-level key like you show, then start Orca, it doesn't see my printer as a "known" one, and when I click on it, it still prompts for the password. After exiting Orac, the config change I made is still there, but sorted in a different place (so it must have been read in and written back out).

rasika-jay, 2024-09-06:
> The workaround for this is to get all your bambu printers connected in Orca, close Orca. Open your `OrcaSlicer.conf` (`~/Library/Application Support/OrcaSlicer/OrcaSlicer.conf` on macOS), find the [...]

This worked for me too

kk7ds, 2024-09-06:
I was eventually able to get it working by putting it in the config, starting orca, selected the printer, put in the code when prompted, then exiting.. Seemed like just putting it in the config was not enough, I had to have it there and then exit orca while that was selected as the active printer or something. After that, it seems to stick now.

Anyway, hopefully Bambu will fix this properly.

forkineye, 2024-09-24:
> The workaround for this is to get all your bambu printers connected in Orca, close Orca. Open your `OrcaSlicer.conf` (`~/Library/Application Support/OrcaSlicer/OrcaSlicer.conf` on macOS), find the [...]

Thank you! This had been driving me nuts...  For those of you that aren't seeing your printer show up under `user_access_code`, connect to your printer on the device tab, then close OrcaSlicer and it should be there. If on Windows, the path is `%appdata%\OrcaSlicer\OrcaSlicer.conf`

adho6298, 2024-11-27:
I'm having the same issue 
Bambu A1 Mini firmware: 01.03.01.02
Orca Slicer: 2.2.0

kk7ds, 2024-11-27:
Still having the same problem as well. The "workaround" mentioned above doesn't work for me more than once. As soon as I power off the printer and re-open Orca, it has forgotten it again. It's super annoying.

I note that the later versions of studio are doing the same thing, so I'm sure it's a bug in the plugin and I'm sure bambu isn't going to fix it. It would sure be nice if Orcaslicer could integrate some sort of workaround, where it re-sets the actual code from a value stored separately or something.

okelk, 2024-11-29:
The workaround mentioned above works for me as long as I only use one printer.
I use 2 bambulabs printers on two different networks.
As soon as use the other printer on the other network orcaslicer has forgotten all the access codes again  😓

f3l1x, 2024-12-26:
Same problem. 

Bambu A1 Mini firmware: 01.04
OrcaSlicer: 2.2.0

flybrys, 2025-01-20:
Workaround kind of works for me but I still need to manually select the printer each time I load Orca

mikeage, 2025-01-20:
The manual selection is probably a different bug -- I see someone opened https://github.com/SoftFever/OrcaSlicer/issues/8097 for it

bowpay, 2025-01-22:
Looks like the program doesn't have access to write to the config file.  Just Run As Administrator (windows) then add the access code and save.  The next time you open it the printer should be good to go.  Worked for me.

flow23, 2025-01-22:
> OK, I have just solved the problem for MacOS 15.1.1 when trying to test the connection and get an error connecting to the printer in the Physical Printer window.
> 
> First, make sure that you only have the most recent version of Orca in your /Applications folder Orca Slicer 2.2.0. I still had a copy of one of the beta releases and that was causing me problems. Next go to System Settings... under the Apple Menu. Look down (or scroll down) and Click on Privacy & Security. Next look down (or scroll down) and click on Local Network. Find Orca Slicer in the list. If Orca Slicer is not enabled, click to enable it. If it is already enabled, click once to disable it and the again to enable it. If you are using Creality Print and having the same problem, make sure Creality Print is enabled as well.
> 
> This resolved all the issues I was having.

This fix doesn't do anything for me. It's random when orcaslicer loses or forgets the connection to the printer.

flybrys, 2025-01-22:
I feel like this is the printer.
I had home assistant and orca working perfectly last night. Closed and reopened several times and it would always connect.

Woke up this morning and everything is disconnected. Restarting the printer, computer, or home assistant didn't fix it.

tueftler0, 2025-01-23:
@flow23 

The correct way is to delete these lower "user_access_code"-section.
The first lines contain an "access_code" section anyway, so Orca will delete the lower ones automatically the next time you start it!

If your printer is not recognized, please take a closer look at the serial number and access code on printer display.
Sometimes a notification window with "Incorrect password" will also appear.

> "access_code": {
> "xxx": "yyy"

xxx = Printer serial number
yyy = Access code

btw.: The now corrupted MD5 hash at the end of the file will also be corrected the next time you start Orca.

tueftler0, 2025-01-23:
> > [@flow23](https://github.com/flow23)
> 
> I can delete the user_access_code section as often as I want. I will be recreated on shutdown and start of OrcaSlicer.

Did you have shut down orca before editing?

flow23, 2025-01-23:
> > > [@flow23](https://github.com/flow23)
> > 
> > 
> > I can delete the user_access_code section as often as I want. I will be recreated on shutdown and start of OrcaSlicer.
> 
> Did you have shut down orca before editing?

Yes

tueftler0, 2025-01-23:
> > > > [@flow23](https://github.com/flow23)
> Yes

Thats quite strange.
I double checked and this section is not regenerated after deletion.

kk7ds, 2025-01-23:
For me, it happens when the printer disappears from the network while orca is open. The computer going to sleep with orca open also triggers it because orca stops being able to access the printer on the network. If I always close orca before sleep or before turning off the printer, it's fine. Every (single) time the printer becomes unreachable from orca, it destroys the linkage and I have to put the code in again.

No amount of editing the conf file ahead of time (yes always with orca closed) will cause it to not do that. If I edit the config before opening orca with the printer running, starting orca will find it. If the printer is offlined with orca open, it breaks the link.

It's insanely frustrating.

Coninox, 2025-01-24:
> For me, it happens when the printer disappears from the network while orca is open. The computer going to sleep with orca open also triggers it because orca stops being able to access the printer on the network.

I can confirm I have this problem with a P1S on LAN mode, and a laptop on Windows. Every time my laptop goes to sleep, Orka ask me the access code.

Wich can appen multiple times a day. It quickly becomes very annoying

kazolar, 2025-01-24:
Solution is find the OrcaSlicer.conf -- find access_code -- you will notice it has a \n at the end of the code, get rid of the \n -- it should be just the code, then the problem will go away

kk7ds, 2025-01-24:
> Solution is find the OrcaSlicer.conf -- find access_code -- you will notice it has a \n at the end of the code, get rid of the \n -- it should be just the code, then the problem will go away

If you mean inside the quoted string of the code, that's definitely not the case in my config. And if you mean outside the quoted string, that makes no sense, since the JSON deserializer wouldn't care about any whitespace.

tueftler0, 2025-01-24:
I use hibernation every day and have not had any problems with this since deleting the user area.
Maybe it would be an idea to write a short script that corrects the file before Orca starts.
Maybe that would be a working solution for you.

kk7ds, 2025-01-24:
Yeah, the problem is that if the computer goes to sleep with orca running (which it does several times during a print) a script modifying orca's config won't have any effect. Which means I'd have to close and re-open orca to notice the config change, which means I'd have to save my project (which I rarely do). This should just work the way it used it - clearly there's some regression here that needs fixing.

tueftler0, 2025-01-24:
You're right - thats bad. I thought that would be a chance for now...

kazolar, 2025-01-24:
> If you mean inside the quoted string of the code, that's definitely not the case in my config. And if you mean outside the quoted string, that makes no sense, since the JSON deserializer wouldn't [...]

yes, inside the quoted string. The "\n" was inside the quoted string. Fixing that has cleared the issue for me

Coninox, 2025-01-24:
> yes, inside the quoted string. The "\n" was inside the quoted string. Fixing that has cleared the issue for me

When the problem appear, the entry "access_code" is not here. It's removed by Orca. (and when it's here, there is no \n)

kazolar, 2025-01-24:
after I cleaned up the config -- i haven't had an issue (it's been just one day -- I'm using 2.2.

christopher-lambe, 2025-01-25:
I have this issue too. Removed Bambu Studio and enabled LAN mode on my P1S. Removed the bambu %appdata% sutff too. Deleted my OrcaSlicer.conf and BambuNetworkEngine.conf too.

OrcaSlicer 2.2 somehow forgets the printer while OrcaSlicer is running and on close removes the access_code or user_access_code entry from the OrcaSlicer.conf file. I saw it live in notepad++ - the file has changed wanna view changes? Yes and then the access_code or user_access_code entry was gone.

TinyWorkshopDesign, 2025-01-25:
mee too same behavior

olsonjj, 2025-01-25:
This is extremely annoying and I'm sure others are having this issue more now since many of us are shifting our printers to LAN mode due to recent firmware news.   

I've read elsewhere this seems to be related to the Bambu network plugin. Its present in both OrcaSlicer and Bambu Studio.  

Perhaps @SoftFever can code around a better fix?

orthadoxtesla, 2025-01-25:
i haven't updated my firmware on the A1 in months but it is happening to me as well and I'm hoping theres a solution soon. i had thought it might just be my crappy internet but it would appear not. any word from Orca Devs? I just moved over to orca and i like the interface but this is rather annoying.

cswingler, 2025-01-26:
I diffed `BambuStudio.conf` against `OrcaStudio.conf`, which revealed that this block belongs at the top level of the JSON object, _outside_ of `app` or other elements.

At the top of OrcaStudio.conf, after the first `{`, just add a block like this:

```
{
    "access_code": {
        "<serial number>": "<access code>"
    },
    "app": {
<snip>
```

I put my serial number and access code there (actually just copy/pasted it from `BambuStudio.conf`, and OrcaSlicer quit bothering me for my access code each time.

kk7ds, 2025-01-27:
> I diffed `BambuStudio.conf` against `OrcaStudio.conf`, which revealed that this block belongs at the top level of the JSON object, _outside_ of `app` or other elements.
>
> At the top of [...]

This is exactly where mine is. If it's there before Orca starts, then it works. If the computer goes to sleep or the printer is taken offline while orca is running, it forgets it again.

I really (really) don't think there's any amount of hacking the config file outside of Orca that is going to fix this, because the access code will be forgotten *while* Orca is running if the printer is unreachable on the network for any amount of time.

TinyWorkshopDesign, 2025-01-27:
> This is exactly where mine is. If it's there before Orca starts, then it works. If the computer goes to sleep or the printer is taken offline while orca is running, it forgets it again.
>
> I really [...]

Worked for me, 
in fact the string was already preset but with another (old) access code

grahamb, 2025-01-27:
I'm seeing this myself; macOS 15.2, OrcaSlicer 2.2.0. P1S in LAN mode. My `user_access_code` is in the top-level of the JSON, and contains no newline characters.

[image]

spali, 2025-01-27:
Had the same problem, but in my case even added multiple times via add printer dialog, it contained a wrong access code in the json. So editing the json manually and enter the correct access code fixed it for me (at least today no problems).

TinyWorkshopDesign, 2025-01-28:
Update: after changing the code at the beginning of the file, the problem appear less frequently, but still appear from time to time...

KatieCakezz, 2025-01-28:
> I diffed `BambuStudio.conf` against `OrcaStudio.conf`, which revealed that this block belongs at the top level of the JSON object, _outside_ of `app` or other elements.
>
> At the top of [...]

on macOS 15.2 I did this but also deleted the access section completely at the bottom of orca config. When I opened orca it gave me an error about it having to fix the config for me, had me set up the first part of the program as if it was freshly downloaded, but as soon as it was done, my printer was already there and I no longer have to re enter the code every time I open orca.

kk7ds, 2025-01-28:
I think most people are experiencing temporary success, like the first start after changing the config, or something. Here's my definitive steps for reproduction:

0. Close Orca
1. Delete all the Orca config stuff so it's a fresh install, or modify the config all you want
2. Start Orca, find the printer, log in with the code if necessary
3. With Orca open and connected to the printer on the device tab, turn off wifi on your computer *or* the printer so it disappears from the network
4. Wait until Orca sees it as disconnected
5. Close and re-open Orca
6. Reconnect wifi, power on the printer, etc
7. When the printer shows up in the left-hand side of the Device tab, it's logged out (i.e. under "Other Device") and will require a code when I click on it

Every. Single. Time. Config. No config. Mangled Config. Praying to the unicorn gods. Doesn't matter.

KatieCakezz, 2025-01-28:
> I think most people are experiencing temporary success, like the first start after changing the config, or something. Here's my definitive steps for reproduction:
>
> 0. Close Orca
> 1. Delete all the [...]

yeap. temporary for sure. I got about an hour and 3 projects before it started doing it again. I'm mad at the unicorn gods for not answering us.

Noisyfox, 2025-01-31:
Please try this PR https://github.com/SoftFever/OrcaSlicer/pull/8256

TinyWorkshopDesign, 2025-01-31:
Tried via Tailscale with an x1c,
seem not Working 

only viable solution for me right now is this python script:

https://gist.githubusercontent.com/Alex-Schaefer/72a9e2491a42da2ef99fb87601955cc3/raw/0bc2423a099ebde636b4067c4cb7316df2c9fa9c/bambu-ssdp-discovery.py￼
bambu-ssdp-discovery
Text Document · 4 KB

spali, 2025-01-31:
> Tried via Tailscale with an x1c,
> seem not Working
>
> only viable solution for me right now is this python script: [...]

Not 100% sure about tailscale... and never checked how bambu does it exactly.
But usually most VPN's do not forward broadcast especially not SSDP which is most of the time only broadcasts to site-local address (239.255.255.250). So your script does probably the same as the printer, but in the name of the printer in the local network.
So this is probably not solvable by orca. Except providing an option to add printer by IP address directly.

Noisyfox, 2025-02-03:
> So the issue I believe is OrcaSlicer erroneously removing the passcode from the config file. It seems if any connectivity problems occur while orcaslicer is open then it thinks there is a problem and then removes it from the config file. Its probably a safety mechanism to not hammer a device with an incorrect pw, but it is overly aggressive in situations where an intermittant network issue may occur. I would suggest that the app be less aggressive in removing, _or_ retain the previous pw in a separate location in the config file and auto populate it back when doing a connect to a "new" device.

yes

[8 more replies]

closed 2025-02-05 by Noisyfox