OrcaSlicer/OrcaSlicer #6177: Arachne causing facets on circular features

OsmosisD, 2024-07-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: AMD Ryzen 7 5800X
RAM: 64Gb DDR4
GPU: NVIDIA GeForce GTX 1650 SUPER

### Printer

Creality CR-100 (Generic Marlin Printer)

### How to reproduce

Slice any circular object using Arachne as the wall generator.

### Actual results

Outer walls come out with distinct facets that also show up on the print. More visible from higher vertical angles.

[image]

This seems to happen regardless of other slicer settings. Increasing resolution to very low numbers does help a little, but increases slicing time significantly.

### Expected results

Arachne should produce smooth(er) walls. The facets still exist when slicing with Classic wall generator, but come out half the length and so are much less noticeable. They also don't have a random offset in the middle.

[image]

### Project file & Debug log uploads

[cylinder.zip](https://github.com/user-attachments/files/16288842/cylinder.zip)
[debug_Thu_Jul_18_16_26_07_23104.log.0.zip](https://github.com/user-attachments/files/16288889/debug_Thu_Jul_18_16_26_07_23104.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

CristianGTR97, 2024-07-25:
I've noticed this too

jwlister, 2024-10-24:
Has there been any progress on this issue? I'm encountering the exact same problem three months later.

closed 2025-01-31 by github-actions[bot]