OrcaSlicer/OrcaSlicer #6182: PETG-HF filament

TinyMito, 2024-07-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Bambu released new PETG-HF filament.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Since I don't have permission to commit, here are the profile files extracted.

[Bambu PETG HF @base.json](https://github.com/user-attachments/files/16304261/Bambu.PETG.HF.%40base.json)
[Bambu PETG HF @BBL A1 0.2 nozzle.json](https://github.com/user-attachments/files/16304262/Bambu.PETG.HF.%40BBL.A1.0.2.nozzle.json)
[Bambu PETG HF @BBL A1 0.8 nozzle.json](https://github.com/user-attachments/files/16304264/Bambu.PETG.HF.%40BBL.A1.0.8.nozzle.json)
[Bambu PETG HF @BBL A1.json](https://github.com/user-attachments/files/16304267/Bambu.PETG.HF.%40BBL.A1.json)
[Bambu PETG HF @BBL A1M 0.2 nozzle.json](https://github.com/user-attachments/files/16304269/Bambu.PETG.HF.%40BBL.A1M.0.2.nozzle.json)
[Bambu PETG HF @BBL A1M 0.8 nozzle.json](https://github.com/user-attachments/files/16304270/Bambu.PETG.HF.%40BBL.A1M.0.8.nozzle.json)
[Bambu PETG HF @BBL A1M.json](https://github.com/user-attachments/files/16304271/Bambu.PETG.HF.%40BBL.A1M.json)
[Bambu PETG HF @BBL X1C 0.2 nozzle.json](https://github.com/user-attachments/files/16304273/Bambu.PETG.HF.%40BBL.X1C.0.2.nozzle.json)
[Bambu PETG HF @BBL X1C 0.8 nozzle.json](https://github.com/user-attachments/files/16304276/Bambu.PETG.HF.%40BBL.X1C.0.8.nozzle.json)
[Bambu PETG HF @BBL X1C.json](https://github.com/user-attachments/files/16304278/Bambu.PETG.HF.%40BBL.X1C.json)

msteele999, 2024-07-19:
Thank you for providing these! I have been downloading the nightly release and don't seem them committed there yet so will test your profiles.

tripp396, 2024-07-24:
How exactly do I import these in Mac. Thought I knew where they went but they are not showing up in the slicer

TinyMito, 2024-07-26:
> How exactly do I import these in Mac. Thought I knew where they went but they are not showing up in the slicer

/Users/***username***/Library/Application Support/BambuStudio/system/BBL/filament

nkamenar, 2024-07-26:
> > How exactly do I import these in Mac. Thought I knew where they went but they are not showing up in the slicer
> 
> 
> 
> /Users/***username***/Library/Application Support/BambuStudio/system/BBL/filament

So interestingly I tried this but they still don't show up in orca slicer after restarting it so I am guessing there is somewhere else that needs updated to tell it about these new filament profiles.

psiberfunk, 2024-07-26:
Per https://github.com/SoftFever/OrcaSlicer/blob/50d646dd41137039712f917aceb43d140e24594c/resources/profiles/BBL.json#L818  my guess is that BBL.json needs to be modified to pick these up ?

raybergh, 2024-07-27:
What and where to put on Windows for 0.2 and 0.4 nozzle?

nkamenar, 2024-07-27:
> Per
> 
> https://github.com/SoftFever/OrcaSlicer/blob/50d646dd41137039712f917aceb43d140e24594c/resources/profiles/BBL.json#L818
> 
> my guess is that BBL.json needs to be modified to pick these up ?

So I tried adding the lines to that file but still doesn't seem to make a difference, still not seeing the new filaments in Orca when clicking the ""Set filaments to use" button.

psiberfunk, 2024-07-27:
Yea I’m not sure how to fix this .. the best thing would be for @Noisyfox or @SoftFever who know the system to integrate the change if just replacing those files doesn’t work .  There must be a change elsewhere

msteele999, 2024-07-27:
Agreed - at this point I am hoping we see it in one of the upcoming nightly builds.

closed 2024-07-30 by SoftFever

msteele999, 2024-07-30:
@SoftFever Will these updates appear in the nightly build for 7/31/2024?

Billy-, 2024-07-31:
> @SoftFever Will these updates appear in the nightly build for 7/31/2024?

According to [the docs](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds), nightly builds are actually built on every commit. I was able to download the latest build from that page, published shortly after this issue was closed, and successfully use PETG-HF filament.

msteele999, 2024-07-31:
> According to [the docs](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds), nightly builds are actually built on every commit. I was able to download the latest build from that [...]

I just downloaded the latest nightly build and confirmed it's there now. Thank you!

dskvr, 2025-04-06:
I'm on 2.3.0 and don't see PETG-HF.

Noisyfox, 2025-04-06:
Have you enabled it?

[image]

commit abb62ca1 (2024-07-30): Update bbl profiles to 01.09.00.18 (#6271)

# Description

Copied files from BambuStudio after downloading updated profiles.

Add support for BambuLab PETG-HF  and Support for ABS. Fixes #6182

Also contains some changes to the default infill from crosshatch to
gyroid, and updated start g-code for A1.

# Tests

Copied all files to resources directory on install, and started.
Profiles showed up.