OrcaSlicer/OrcaSlicer #6186: Proposal for Developing a Mobile Version of Orca Slicer

RoboLask, 2024-07-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Dear Orca Slicer Developers,

I would like to express my appreciation for your product, which significantly simplifies the process of preparing models for 3D printing. However, I would like to propose an idea that I believe could greatly expand your user base and make working with 3D printers more accessible for many people.

The idea is to create a fully-featured version of Orca Slicer for iPhones, iPads, and Android smartphones. Being able to use the slicer directly on mobile devices would be incredibly convenient for many users. Here are a few key advantages such a solution could bring:

1. **Accessibility**: Not everyone has access to personal computers, especially newcomers to 3D printing. Almost everyone, however, has a smartphone, and modern smartphones are powerful enough to handle complex tasks.

2. **Convenience**: The ability to work with models and send them to print directly from a phone would allow users to engage in 3D printing anytime and anywhere they have access to a printer.

3. **Broader Audience**: Developing a mobile version of the slicer would attract new users who previously couldn't start 3D printing due to the lack of a PC.

4. **Modern Technology**: Today's smartphones have processing power comparable to PCs, enabling them to handle even complex tasks.

I am confident that implementing a mobile version of Orca Slicer would be a significant step forward and attract new users to your product, as well as simplify the 3D printing process for existing users. I hope you will consider my suggestion and incorporate it into your product development roadmap.

Thank you for your attention and your hard work!

Best regards,  
Robolask

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like Orca Slicer to develop a fully-featured version of their 3D printing slicer software for mobile devices, specifically for iPhones, iPads, and Android smartphones. This mobile version should include all the essential functionalities currently available in the desktop version, allowing users to prepare, slice, and send models directly to 3D printers from their mobile devices.

Key Features:
Model Import and Manipulation: Users should be able to import 3D models in various formats (e.g., STL, OBJ) and manipulate them by scaling, rotating, and positioning them on the build plate.
Slicing and Preview: The mobile version should offer advanced slicing capabilities, including setting layer height, infill density, print speed, and other critical parameters. A preview function to visualize the sliced model layer-by-layer would be essential.
Printer Compatibility: The app should support connecting to and managing a wide range of 3D printers via Wi-Fi, Bluetooth, or USB OTG (On-The-Go).
Cloud Integration: Integration with cloud storage services to save, retrieve, and share sliced files easily would be highly beneficial.
User-Friendly Interface: An intuitive and responsive user interface designed for touchscreens, making it easy for users to navigate and use all features on smaller screens.
Remote Monitoring and Control: The ability to monitor and control the printing process remotely, including starting, pausing, and stopping prints, as well as viewing live camera feeds of the printing process if supported by the printer.
Offline Mode: The app should be able to function offline, allowing users to slice and prepare models even without an internet connection.
Performance Optimization: Ensure that the app is optimized for mobile hardware to deliver smooth performance and efficient use of battery life.
Benefits:
Accessibility: Making the slicer available on mobile devices will lower the entry barrier for newcomers to 3D printing who may not have access to a PC.
Convenience: Users will be able to prepare and manage their 3D prints on the go, without being tied to a desktop or laptop computer.
Expanded User Base: Attracting a broader audience, including those who primarily use mobile devices, will increase the user base and foster greater interest in 3D printing.
Innovation: Staying ahead in the market by leveraging the processing power of modern smartphones and catering to the growing demand for mobile solutions.
By developing a mobile version of Orca Slicer, you will provide users with unprecedented flexibility and convenience, further establishing Orca Slicer as a leading solution in the 3D printing community.

MagicBusX, 2024-07-19:
Great Idea!

cmiles68, 2024-07-26:
Was just searching the web for the very same thing! 

That would be incredible.

closed 2024-11-01 by github-actions[bot]