OrcaSlicer/OrcaSlicer #6190: "Filament load time"/"Filament unload time" missing from UI

orion23rigel, 2024-07-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron Trident (customized with ERCF)

### How to reproduce

1. Go to the printer settings. 2. Search for "filament load time". 3. Click on the result that appears (indicating the setting is under Advanced). 4. Realize the setting is not there (even with the Advanced toggle activated).

### Actual results

[image]
[image]

The respective settings are not available.

### Expected results

It is possible to configure the filament load and unload times so the slicer time estimate is more accurate.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Previous [issue 3311](https://github.com/SoftFever/OrcaSlicer/issues/3311) has a [comment](https://github.com/SoftFever/OrcaSlicer/issues/3311#issuecomment-1894102852) indicating they might have been hidden on purpose, but there is also indication there that they should be available. Please let me know if this should be a feature request instead of a bug report. Thanks for all your work on this excellent project!

orion23rigel, 2024-07-20:
I manually added `"machine_load_filament_time": "15",` and `"machine_unload_filament_time": "30",` to the json file for the printer, but the time estimate does not change when I change the values there (closing the app, changing the json values and saving, launching the app again, and reloading the printer in the project) and the fields still don't appear in the printer configuration window. OrcaSlicer does recognize that the printer config has changed, as it has the asterisk next to it when I reopen the project.

lhndo, 2024-10-21:
same issue over here

closed 2025-01-28 by github-actions[bot]