OrcaSlicer/OrcaSlicer #6191: Some presets only load every other program start orca 2.1.1

falconlb45, 2024-07-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

	Windows 11 Home 64-bit
CPU
	Intel Processor
	Cores	12
        Threads	20
        Specification	12th Gen Intel Core i7-12700
RAM
	64.0GB
Motherboard
	LENOVO 3741 (U3E1)	
Graphics
	M557-G0 (1280x720@60Hz)
	Intel UHD Graphics 770 (Lenovo)
	4095MB NVIDIA GeForce RTX 3060 (Lenovo)	
	SLI Disabled
Storage
	931GB Western Digital WDC WD10EZEX-08WN4A1 (SATA )	97 °F
	1907GB Timetec MS12 (Unknown (SSD))
	7452GB Western Digital WD easystore 264D USB Device (USB (SATA) )	
	903GB Seagate ST1000LM 035-1RK172 SCSI Disk Device (USB (SATA) )	
Optical Drives
	No optical disk drives
Audio
	NVIDIA High Definition Audio

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

1 start orca
2 new project
3 select preset under user presets

 When I start orca my user presets are not all present. I close the program and start it again and they are all there. I close and start again and the same user presets are missing again.
I have tried uninstalling, reinstalling
uninstalling, installing older version, and then updating
uninstalled with revo uninstaller, restarted pc, ran from portable install on usb drive

### Actual results

1 start orca
2 new project
3 select preset under user presets, but user preset I want is only present on every other boot

what I see 
[image]

what I expect to see
[image]

### Expected results

the user preset to be there everytime the software loads
[Elegoo Neptune 4 plus jerk.zip](https://github.com/user-attachments/files/16318071/Elegoo.Neptune.4.plus.jerk.zip)
this is a copy of the preset that loads every other boot

### Project file & Debug log uploads

There is no project file, the issue occurs when you make a new project
log files 
[log.zip](https://github.com/user-attachments/files/16318091/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

issue occurs exactly every other time i load orca slicer
log says it cant find the presets in appdata but they are there
I have included a zip file containing the preset that does not load every time

falconlb45, 2024-07-20:
i have also posted this issue to discord
https://discord.com/channels/1137181739773603922/1263999728513716315

akmjoe, 2024-07-20:
This issue was also in version 2.0.0 - [issue 5438](https://github.com/SoftFever/OrcaSlicer/issues/5438)

falconlb45, 2024-07-20:
> This issue was also in version 2.0.0 - [issue 5438](https://github.com/SoftFever/OrcaSlicer/issues/5438)

I mistook that issue as none of the user presets load, whereas my issue is that some presets load but others do not.

ufoDziner, 2024-08-09:
I have the same issue with the Mac version.  I did manually copy my user presets over from my windows computer to the mac folder.  Not sure if that makes any difference.

closed 2024-11-15 by github-actions[bot]

CarVac, 2025-01-05:
This persistently happens for me. Every other time I launch OrcaSlicer, some of my machine profiles are absent.

It occurs on 2.2.0 on Linux. I also recall Luke from Luke's Lab saying it happens to him on multiple OS's.

buzzhuzz, 2025-01-11:
@CarVac, could you please collect some debug logs for this problem?

Open your `~/.config/OrcaSlicer/OrcaSlicer.conf` and modify/add "log_severity_level" to trace:
```
"log_severity_level": "trace",
```

After that, cleanup your `~/.config/OrcaSlicer/log` directory and run OrcaSlicer few times to get good and bad cases. Archive and share contents of log folder.

You may return log level back to original value (or remove that line from config to use defaults).

CarVac, 2025-01-15:
[missing_profiles.zip](https://github.com/user-attachments/files/18417732/missing_profiles.zip)
[found_profiles.zip](https://github.com/user-attachments/files/18417734/found_profiles.zip)

@buzzhuzz these are from two consecutive openings of the AppImage.

Here are my most pertinent profiles:

[KLP1 0.5 nozzle.json](https://github.com/user-attachments/files/18417750/KLP1.0.5.nozzle.json)
[0.24mm Standard @KLP1.json](https://github.com/user-attachments/files/18417760/0.24mm.Standard.%40KLP1.json)
[Inland ASA - KLP1.json](https://github.com/user-attachments/files/18417762/Inland.ASA.-.KLP1.json)

buzzhuzz, 2025-01-18:
Thank you for the logs.

Now I see that issue is reproducible on my side too. I don't notice it since I have no any profiles derived from Custom/Generic profiles.

However, every other run Custom profiles are not loaded. Moreover, I see that contents of ~/.config/OrcaSlicer/system alters every run:
1. At first run there are Creality profiles only
2. On second run there are Creality and Custom profiles.
3. Third run - Creality only.
4. and so on

I've added some extra logging into and see that at first run Custom profiles removed and on second run there are added back.

@Noisyfox , there are clearly some problem happens in profiles updater. Is it possible to reopen this issue or we need to create a new one?

EFlynn7778, 2025-01-19:
I am having this issue as well. Its been happening for a while with several versions on Windows 10 and 11.  I recently updated to 2.2.0 to see if it would fix the issue. It did not and is still present.

Just as described above.  Every other run it does not load the user presets.

darkytoothpaste, 2025-01-21:
I have the same problem too.