OrcaSlicer/OrcaSlicer #6194: Add parameters for MMU

syminomega, 2024-07-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I've built the ERCF and installed the happy hare on my printer. However, I was stacked when i was setting up in the slicer. The MMU needs to setup some parameters before printing, such as `REFERENCED_TOOLS`, `PURGE_VOLUMES`. I could not found these parameters in Orca Slicer.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I hope if we could add these g-code placeholders:
REFERENCED_TOOLS: The index list of tools which were used. `0,2,5,6` means that T0, T2, T5 and T6 are used in the print.
PURGE_VOLUMES: This placeholder is substituted with a comma separated list of NxN purge volumes.
NEXT_FLUSH_VOLUME: How much material to flush after tool change.

igiannakas, 2024-07-20:
They do exist - copy paste the setup scrips from the HH slider tuning guide. https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup

igiannakas, 2024-07-20:
You need object processing enabled in your klipper install for these to work.

syminomega, 2024-08-11:
> They do exist - copy paste the setup scrips from the HH slider tuning guide. https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup

Thanks, it works. I notice that the slicer does not provide these parameters directly.

closed 2024-08-11 by syminomega