OrcaSlicer/OrcaSlicer #6211: Gaps between Infill and Perimeters & Sparse Infill Where it Doesn't Belong

CCS86, 2024-07-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

10 x64

### Printer

X1C

### How to reproduce

...

### Actual results

For some reason, I am getting gaps between the infill and perimeter walls, plus one layer has an odd section where it is putting sparse infill where it doesn't belong. 

### Expected results

https://github.com/user-attachments/assets/1f48699f-1cb0-4f4c-b77a-fd29b18b4cb8

### Project file & Debug log uploads

[debug_Sun_Jul_21_17_17_22_424128.log.zip](https://github.com/user-attachments/files/16326252/debug_Sun_Jul_21_17_17_22_424128.log.zip)

[project.zip](https://github.com/user-attachments/files/16326254/project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

thebandkid, 2024-07-25:
try lowering with your line width or just slowing down your print

CCS86, 2024-07-26:
> try lowering with your line width or just slowing down your print

Why would I do either of those things? 

This is a bug in the slicing engine.

closed 2025-11-02 by github-actions[bot]