OrcaSlicer/OrcaSlicer #6227: Wrong temperatur at TPU temperatur calibration

Daniel-0187, 2024-07-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron Trident 250mm

### How to reproduce

Calibration->Temperatur->TPU

### Actual results

[image]
Start-Temp. 320°C

### Expected results

Start-Temp. 230°C, i think the numbers are in the wrong order.

### Project file & Debug log uploads

[Flowrate Test-Pass1.zip](https://github.com/user-attachments/files/16346347/Flowrate.Test-Pass1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-07-24:
320° ?!?!?!?!?  That's crazy.  For TPU - Not on your life.
You're probably getting the wrong temperature on the machine because that crazy high it probably out of range for the hotend and the machine firmware won't allow it.

GameOEver, 2024-07-24:
I just found the same issue.

My suggestion is to lower both temps by 100°C. 220 to 180°C sounds good to me.

sasha-feig, 2024-07-28:
I got the same issue, P1S. (tho I guess the default values are the same for everything)

closed 2024-11-04 by github-actions[bot]