OrcaSlicer/OrcaSlicer #6228: Fuzzy Skin is not random

Robsbiche, 2024-07-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

BambuKLa

### How to reproduce

Apply fuzzy skin only on a layer range (my case 2mm to 9.5mm)
Set the fuzzy skin setting to outer perimeter only, 0.2 distance and 0.2 thickness

### Actual results

[image]

### Expected results

The fuzzy skin displays a repetitive pattern instead of a random fuzzy surface

### Project file & Debug log uploads

[Fuzzy skin issue.zip](https://github.com/user-attachments/files/16350323/Fuzzy.skin.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

L2on1, 2024-07-24:
I thought it was fixed #5682

Robsbiche, 2024-07-24:
> I thought it was fixed #5682

It seems like it was, but the problem is still existing in my case.
Maybe because I only apply fuzzy skin on a selective layer height range.

discip, 2024-07-26:
@Robsbiche
Your project with the latest build:
[image]
🤷‍♂️

Robsbiche, 2024-07-26:
> @Robsbiche Your project with the latest build: [...]

Well it's even weirder then.
I use the 2.1.1, isn't it the latest build?

Noisyfox, 2024-09-15:
Arachne is still broken. The PR #5682 only fixes for classic mode.

Noisyfox, 2024-09-15:
> @Robsbiche Your project with the latest build: [...]

This bug mostly affects Windows user, due to how the `rand()` function is implemented in MSVC.

PR #6770 (2024-09-15): Fix fuzzy skin randomness for Arachne wall generator
Fix #6228 by applying the same fix as #5682 to Arachne wall generator.

Current:
[image]

Fixed:
[image]
closed 2024-09-15 by SoftFever