OrcaSlicer/OrcaSlicer #6229: Creality K1/K1C - the exhaust fan does not turn on

allexmail, 2024-07-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality K1/K1C

### How to reproduce

1. Turn on the exhaust fan in the profile settings.
2. Generate a gcode.

### Actual results

The exhaust fan does not react (does not turn on).

### Expected results

Turning on the exhaust fan according to the set settings.

### Project file & Debug log uploads

[K1c_project_and_log.zip](https://github.com/user-attachments/files/16350386/K1c_project_and_log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

This issue has[ already been raised here](https://github.com/SoftFever/OrcaSlicer/issues/2183), but it is closed without any comments. The problem is that in the settings Orca uses the code "M106 P3 S123", however, in the latest firmware of the printer V1.3.3.29 (+ root + Fluidd) the numbering is used as follows:

P0 = Print head fan
**P1** = Housing fan (exhaust fan)
P2 = Side fan

Of course, instead of the correct numbering P1, the number P3 is inserted into the gcod to turn on the exhaust fan. Therefore, the fan does not start.
I do not know if it is possible to change the fan numbering in the printer itself, so I am writing here. Perhaps in orca you can make parameters for numbering fans.

bistory, 2024-07-24:
This issue is due to Creality not using correct conventions, an issue has been opened here but is ignored by them : https://github.com/CrealityOfficial/K1_Series_Klipper/pull/7

BluBb384, 2024-07-26:
Hello, to fix the problem I use filament start/end gcode

m106 p1 s255
m106 p1 s0

closed 2024-11-01 by github-actions[bot]