OrcaSlicer/OrcaSlicer #6232: CalibPressureAdvancePattern does z-hop before retraction

gregmatic, 2024-07-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

10

### Additional system information

N/A

### Printer

Ender 3

### How to reproduce

- Use built-in Pressure Advance: PA Pattern Tool. any setting will do
- Slice and inspect Gcode at retraction at the end of each chevron. e.g top layer

### Actual results

1. z-hop of 0.2mm is hardcoded before every travel move between a completed chevron and the next chevron to be printed, and cannot be disabled or modified by any associated setting
2. if retraction is enabled, the hardcoded z-hop is inserted after the retraction instead of before. i.e. z-hop->retract->move.  this provokes unnecessary stringing in the PA Pattern Cal because it creates a vertical string/ooze at the end of a chevron and then drags it across to the other side where it is deposited on previously completed chevrons and/or the start of the new chevron.
3. this is opposite of how z-hop is typically executed in the slicer outside of PA Pattern Cal, i.e. retract->z-hop->move.
4. below i have 3 examples of the same PA Pattern config. 

4A. First on the left, Orca PA Pattern Cal wizard with hardcoded z-hop and ooze/blobs all over the place. 
4B. Second in the middle from [Ellis Pattern Generator](https://ellis3dp.com/Pressure_Linear_Advance_Tool/) with z-hop enabled and set to 0.2mm (the hardcoded value in Orca PAPC), with reduced ooze/blobs. This is because in Ellis Generator, the z-hop is performed before the retraction.
4C. Third on the right is also Ellis Generator but with z-hop disabled. Significantly less stringing and very minor zits deposited randomly

 [image]
5. In OrcaSlicer/src/libslic3r/calib.cpp the z-hop is hardcoded before every travel move and undo z-hop after. this block is found several times in void CalibPressureAdvancePattern::generate_custom_gcodes()
```
- gcode << m_writer.travel_to_z(zhop_height, "z-hop before move");
- gcode << move_to(Vec2d(to_x, to_y), m_writer, "Move to pattern start");
- gcode << m_writer.travel_to_z(layer_height, "undo z-hop");
```
6. But retraction is already inside the move_to function, which means it gets priority to be next the move_to command

```
std::string CalibPressureAdvance::move_to(Vec2d pt, GCodeWriter &writer, std::string comment)
{
    std::stringstream gcode;

    gcode << writer.retract();
    gcode << writer.travel_to_xy(pt, comment);
    gcode << writer.unretract();

    m_last_pos = Vec3d(pt.x(), pt.y(), 0);

    return gcode.str();
}
```
7. 5. & 6. result in z-hop being done before retraction.

### Expected results

1. z-hop should obey its associated setting in the slicer. i.e. allow enable/disable, allow height adjustment of z-hop 
2. then move all the z-hop and undo z-hops into std::string CalibPressureAdvance::move_to() 

```
std::string CalibPressureAdvance::move_to(Vec2d pt, GCodeWriter &writer, std::string comment)
{
    std::stringstream gcode;
    gcode << m_writer.travel_to_z(zhop_height, "z-hop before move");
    gcode << writer.retract();
    gcode << writer.travel_to_xy(pt, comment);
    gcode << writer.unretract();
    gcode << m_writer.travel_to_z(layer_height, "undo z-hop");
    m_last_pos = Vec3d(pt.x(), pt.y(), 0);

    return gcode.str();
}
```
3. then all of this
```
- gcode << m_writer.travel_to_z(zhop_height, "z-hop before move");
- gcode << move_to(Vec2d(to_x, to_y), m_writer, "Move to pattern start");
- gcode << m_writer.travel_to_z(layer_height, "undo z-hop");
```
can be replaced with this
```
gcode << move_to(Vec2d(to_x, to_y), m_writer, "Move to pattern start");
```

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ergonomicmike, 2024-07-24:
I concur that the Orca version of the Ellis pattern isn't as good as an original Ellis pattern.

I don't have a photo, but I tried the Orca PA pattern last week. Twice. Both prints tore apart on my printer. Whereas the original Ellis always printed well.

igiannakas, 2024-07-25:
I've pushed a PR with a fix for this here: https://github.com/SoftFever/OrcaSlicer/pull/6245

You can download and test it here:
https://github.com/SoftFever/OrcaSlicer/actions/runs/10091976362

Feedback appreciated :)

igiannakas, 2024-07-25:
@gregmatic As you're testing the adaptive PA feature, and since we don't merge PR's into each other for code management sake, you can download a consolidated build including both the above PA pattern z hop and retraction fix plus the adaptive PA (and other enhancements that are currently PR's that I've consolidated for my day to day use) from my dev branch here: https://github.com/igiannakas/OrcaSlicer/actions/runs/10092022209

gregmatic, 2024-07-29:
> @gregmatic As you're testing the adaptive PA feature, and since we don't merge PR's into each other for code management sake, you can download a consolidated build including both the above PA [...]

really appreciate it man!

igiannakas, 2024-07-29:
Adaptive PA is now merged in the nightly so you can just download the action build from the PR above ;)

PR #6245 (2024-08-06): Fix z hop performed after retraction in PA pattern calibration. Fix z hop config value not used
# Description

Fix z hop performed after retraction in PA pattern calibration. Fix z hop extruder config value not used

Fixes #6232

## Screenshots:
0.4 z hop
[image]

0 zhop
[image]

## Tests

Checked with a z hop value set in the extruder. Z hop operations performed correctly (with the right height) and after the retraction operation. 
Checked restoring of z height after the travel move. Works ok. Checked de-retraction after travel move. Works OK
Checked with a 0 z hop height in the extruder sect [... cut]
closed 2024-08-06 by SoftFever