OrcaSlicer/OrcaSlicer #6233: Total Estimation overview

zsculpt, 2024-07-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It drives me insane that I can't see the total time estimation when I'm looking at any other views besides "Line Type", in particular the "Speed" and "Layer Time" views. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please make the total estimation sub-view available to all preview modes (Speed, Layer Time, etc.) instead of only being visible under "Line Type"

DVSVIDEO, 2024-07-25:
Right! That annoys me too...

kylek29, 2024-07-26:
Agreed. Ideally it'd have a collapsible panel of details that is separate from the non-view specific stats.

It'd be nice to have it do breakdown by plate and objects on that plate.

PR #10333 (2025-08-24): GCodeViewer will now always show the estimations
Closes: #6233 
Closes: #8717 

The preview will now always display time estimations, regardless of the selected display type. For example, if the display type is set to “Flow,” the preview will still show the estimated times. 

[image]