OrcaSlicer/OrcaSlicer #6242: white (invisible) font issue on linux MATE desktop

sercona, 2024-07-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.0.0 app image for linux

### Operating System (OS)

Linux

### OS Version

mint 20.3

### Printer

sovol sv06

### How to reproduce

do a calibration or any operation where a dialog box pops up to show a changed profile value (old and new value).  the text is white on white and nearly impossible to see.  I know this has been reported several times before but there has been no solution and any linux user running this is likely to encounter it.

### Actual results

white text on white bg on dialog boxes

### Expected results

readable non-white text!

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sercona, 2024-10-23:
is there any work-around?  not being able to see the text is kind of important, no?

closed 2025-01-31 by github-actions[bot]

EricBellamy, 2025-02-14:
Getting this too, light mode Mac OSX is unusable. Dark mode is unusable but you can at least tell text exists.