OrcaSlicer/OrcaSlicer #6246: No Filament Option for Dremel Printer available

Reevoy24, 2024-07-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Dremel 3D45

### How to reproduce

If you add a new Printer and select any Dremel Printer and then go to the add Fillament Option you cant add Filament the Window is empty.
[image]

### Actual results

you cant add a prefert filament and you also cant add a custom filament

### Expected results

you should see a lot of pre designt filament options like PLA or PETG

### Project file & Debug log uploads

i dont get an error code so there is no log to look at

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Reevoy24, 2024-07-28:
This is the only log i have
[debug_Sun_Jul_28_15_40_57_21256.log](https://github.com/user-attachments/files/16403920/debug_Sun_Jul_28_15_40_57_21256.log)

cochcoder, 2024-09-27:
Hi, this should now be fixed

closed 2025-01-03 by github-actions[bot]