OrcaSlicer/OrcaSlicer #6251: Part not printing as sliced

aideym, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11 pro

### Additional system information

Intel i7 8850h Nvidia Quadro Pro

### Printer

Bambulabs X1c

### How to reproduce

3 parts of 1 assembly imported to slicer
Sliced and sent to printer
2 parts print as expected 
1 part is dimensionally incorrect

### Actual results

[image]
[image]
[image]

### Expected results

All three expected to be ptinted as sliced. The incorrect item is Part studio 1 - retainer

### Project file & Debug log uploads

[Part Studio 1 - pusher.zip](https://github.com/user-attachments/files/16387787/Part.Studio.1.-.pusher.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I sliced the same file in Bambuslicer using the same settings and got the expected result

ItsJupiter21, 2024-07-28:
can you share the project 3mf and the onshape link/the step file?

aideym, 2024-07-28:
Of Course:
https://cad.onshape.com/documents/47ef673d6732d8ce394de945/w/15f495752a1eb6dbbeb95b4f/e/5b7019d8bb341658c566e81a?renderMode=0&uiState=66a6452f0c9a7537dc971f88

[Part Studio 1(2).zip](https://github.com/user-attachments/files/16403873/Part.Studio.1.2.zip)

closed 2024-11-04 by github-actions[bot]