OrcaSlicer/OrcaSlicer #6254: out of range error after using baby stepping

moth4017, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

windows 64 pro

### Printer

Valkyrie HT

### How to reproduce

start a prin and they use the "baby stepping" to set z height anything grater than -0.2 i get this error
<html>
<body>

PM | Error: G3: intermediate position outside machine limits Cancelled printing file 0:/gcodes/inner (1)_PLA_32m19s.gcode, print time was 0h 3m Error: in GCode file line 187: G3: outside machine limits
-- | --

</body>
</html>

### Actual results

<html>
<body>

PM | Error: G3: intermediate position outside machine limits Cancelled printing file 0:/gcodes/inner (1)_PLA_32m19s.gcode, print time was 0h 3m Error: in GCode file line 187: G3: outside machine limits
-- | --

</body>
</html>

g code attached 

### Expected results

should be able to bay step greater than -0.25

### Project file & Debug log uploads

[Uploading inner (1)_PLA_32m19s.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

Ergonomicmike, 2024-07-26:
On the assumption that your printer uses Klipper, I politely suggest that this is not an "Issue" in Orca. Rather, it is a problem with your printer.cfg in Klipper. I say this for two reasons: 

1) When I first started using Klipper I had received this same error when baby stepping my Z-height. I had to change `position_min` in printer.cfg. (In my case, I changed it to -8.0.) 

2) Orca doesn't know anything about baby stepping your printer. Orca simply generates G-code.

moth4017, 2024-07-27:
Hi , not using klipper, running RRF

Regards

Mark

closed 2024-11-03 by github-actions[bot]