OrcaSlicer/OrcaSlicer #6259: Tree support on big prints print internal floating in air things inside support

xTremePower, 2024-07-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11

### Printer

Elegoo Neptune 4 Max

### How to reproduce

Slice a big model and see internal volumes of the tree supports

### Actual results

The slicer generate floating tree parts, totally non sence in mid air or even more annoying, reduce the tree branch diamter directly and print in mid air (see pics)

### Expected results

A tree is not supposed to print with nothing under, and I have no way to reduce this, if I lower the branche angle, with a 400mm print, tree will become weak
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Project file & Debug log uploads

not needed, make this with any big parts

### Checklist of files to include

- [ ] Log file
- [ ] Project file

xTremePower, 2024-07-26:
[image]

empirebuilder1, 2024-07-30:
also having this issue. Tree supports seem to have no enforcement for Z-axis layer continuity. When printing parts with small overhangs that then have a tree above it, it seems to "shift" the tree over causing a print-in-air problem and other nonsensical geometry. 

Example below, showing a part that has a horizontal lip that the tree must then support above- the layer shift occurs slightly before when the tree tries to "shift over" to accomodate the lip. 
[https://i.imgur.com/3EH5HRw.png](url)

Files attached exhibiting the problem.
[90pipe_WINDOWtoHOSE_V3_SlicedFilesWithError.zip](https://github.com/user-attachments/files/16423074/90pipe_WINDOWtoHOSE_V3_SlicedFilesWithError.zip)

closed 2024-11-06 by github-actions[bot]